realsenseai/librealsense #728: Build failure

wexi, 2017-10-20:
enoch@dome:~/librealsense/build$ uname -a
Linux dome 4.13.0-16-generic #19-Ubuntu SMP Wed Oct 11 18:35:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Commit 67cdafef

CMakeFiles/realsense2.dir/build.make:326: recipe for target 'CMakeFiles/realsense2.dir/src/error-handling.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/error-handling.cpp.o] Error 1

dorodnic, 2017-10-21:
Hi @wexi 
The build is [passing](https://travis-ci.org/IntelRealSense/librealsense). If your application depends on previous version of librealsense, please configure it to track [the legacy branch](https://github.com/IntelRealSense/librealsense/tree/legacy) instead of master.

wexi, 2017-10-22:
It's my first build attempt and it is failing. Here is how:

CMakeFiles/realsense2.dir/build.make:326: recipe for target 'CMakeFiles/realsense2.dir/src/error-handling.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/error-handling.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Line 326 of the a/m makefile is:
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/home/enoch/librealsense/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_12) "Building CXX object CMakeFiles/realsense2.dir/src/error-handling.cpp.o"
	/usr/bin/c++  $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -o CMakeFiles/realsense2.dir/src/error-handling.cpp.o -c /home/enoch/librealsense/src/error-handling.cpp

This error-handling.cpp is not missing. How can I turn on debug logging of the build process?

freemanlo, 2017-10-23:
Hi @wexi 

The librealsense2 (tag: 2.8.0) build should be fine. My build machine is Ubuntu 16.04 + kernel 4.4.0.97. Please check below instructions to see if you see the same issue:
https://github.com/IntelRealSense/librealsense/blob/v2.8.0/doc/installation.md

wexi, 2017-10-23:
Hi, Unfortunately going back to "2.8.0" did not help, same build error and same error going forward with the "development" branch:

CMakeFiles/realsense2.dir/build.make:326: recipe for target 'CMakeFiles/realsense2.dir/src/error-handling.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/error-handling.cpp.o] Error 1

I am not experienced with cmake. Is there a standard way to tell it to build a Makefile with a more meaningful debug output?

uname -a
Linux dome 4.13.0-16-generic #19-Ubuntu SMP Wed Oct 11 18:35:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

ev-mp, 2017-10-24:
Hi@ wexi,
What is the command line params you use to invoke cmake and make, and what are the outputs.
Please provide the full log of the two

wexi, 2017-10-25:
master branch now builds fine. Thanks.

closed 2017-10-25 by wexi