realsenseai/librealsense #729: Problem with OpenMP dependency on macOS

auni53, 2017-10-20:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |  3.21 | 
| Operating System & Version | macOS 10.12.4   |
| Kernel Version (Linux Only)  |   |

I call `make ../ -DBUILD_EXAMPLES=true` and get the following: 
> cmake ../ -DBUILD_EXAMPLES=true                                                                                                            ~/Workspace/librealsense/build(master)@Aunis-MacBook-Pro.local
-- Info: REALSENSE_VERSION_STRING=2.8.0
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES) (found version "1.0")
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES) (found version "1.0")
CMake Error at CMakeLists.txt:58 (message):
   OpenMP package is missing!
-- Configuring incomplete, errors occurred!
See also "/Users/Auni/Workspace/librealsense/build/CMakeFiles/CMakeOutput.log".

This happened before when I briefly tried the development branch. I think it has to do with OpenMP being moved to llvm in brew, similar to [here](https://github.com/Microsoft/LightGBM/issues/118).

dorodnic, 2017-10-21:
Hi @auni53 
You can disable OpenMP in CMake using `-DBUILD_WITH_OPENMP=false`. 
However, D400 support for Mac OS is still work-in-progress (the D400 is using UVC 1.5 spec, that requires considerable modifications to the existing `libuvc` backend). 
For other devices, we recommend using [the legacy driver](https://github.com/IntelRealSense/librealsense/tree/legacy).

auni53, 2017-10-21:
Fantastic, just figured out that flag was a thing myself haha. Hope this helps anyone else having issues messing around with clang and llvm on mac!

closed 2017-10-21 by auni53

svenevs, 2017-10-26:
I needed to prepare instructions for somebody on how to get OpenMP support with this, the unofficial guide is [here](https://gist.github.com/svenevs/f00ed3898d2af6248921b63254aa8cc1).