realsenseai/librealsense #730: can´t find librealsense2/rsutils.hpp

sglxh, 2017-10-21:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  ubuntu14.04|
| Kernel Version (Linux Only)  |   4.4.0-97-generic|

I want to Displaying Infrared Frame 
https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md  .
After I installed the Intel SR300 driver, I find it need a hpp file  <librealsense2/rsutils.hpp>.I can find rs.hpp but I can´t find rsutils.hpp at /usr/local/include/librealsense2.  Where is the rsutils.hpp?

dorodnic, 2017-10-21:
Hi @sglxh 
I see now that the documentation is somewhat out of date, it will be updated shortly.

zivsha, 2017-10-21:
Hi @sglxh,

Until we update the sample, you can try the following snippet (which does not require `rsutils.hpp`:
```cpp
    rs2::pipeline pipe;  //Contruct a pipeline which abstracts the device
    rs2::config cfg;   //Create a configuration for configuring the pipeline with a non default profile

    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_Y8, 30); 
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

    //Instruct pipeline to start streaming with the requested configuration:
    pipe.start(cfg);

    //Try to get an infrared frame
    rs2::frame ir_frame;
    while(!ir_frame)
    {
        ir_frame = pipe.wait_for_frames().first(RS2_STREAM_INFRARED);
    }
    
    //Use infrared frame...
    //Same code as the original example here...
    Mat ir(Size(640, 480), CV_8UC1, (void*)ir_frame.get_data(), Mat::AUTO_STEP);
    /....
```

ankit-shinde, 2017-10-26:
Hi zivsha,
I have tried your snippet but getting following error:

terminate called after throwing an instance of 'rs2::error'
 what():  Frame of requested stream type was not found!
Makefile:10: recipe for target 'run' failed
make: *** [run] Aborted (core dumped)

System info:

Required Info |  
-- | --
Camera Model | SR300
Firmware Version |  
Operating System & Version | ubuntu16.04
Kernel Version (Linux Only) | 4.12.12-041212-generic

zivsha, 2017-10-26:
Follow up on Issue #742

closed 2017-11-01 by dorodnic