realsenseai/librealsense #742: Opencv new updates in pull request

ankit-shinde, 2017-10-26:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Linux 16.04 LTS  |
| Kernel Version (Linux Only)  | 4.12.12  |

// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

// include OpenCV header file
#include <opencv2/opencv.hpp>

using namespace std;
using namespace cv;

int main()
{
        //Contruct a pipeline which abstracts the device
        rs2::pipeline pipe;

        //Create a configuration for configuring the pipeline with a non default profile
        rs2::config cfg;

        //Add desired streams to configuration
        cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);

        //Instruct pipeline to start streaming with the requested configuration
        pipe.start(cfg);

        // Camera warmup - dropping several first frames to let auto-exposure stabilize
        rs2::frameset frames;

        for(int i = 0; i < 30; i++){
            //Wait for all configured streams to produce a frame
            frames = pipe.wait_for_frames();
        }

        //Get each frame
        rs2::frame color_frame = frames.get_color_frame();

        Mat color(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
     
        namedWindow("Display Image", WINDOW_AUTOSIZE );
        imshow("Display Image", color);

        waitKey(1);
    
        return 0;
}

Compiles successfully, but segmentation fault when run. How to display image in opencv

zivsha, 2017-10-26:
Where do you get the segmentation fault?

ankit-shinde, 2017-10-26:
When running the code simply getting error:
Segmentation fault (core dumped)
After using gdb found:
Thread 1 "main" received signal SIGSEGV, Segmentation fault.
0x00007ffff3713298 in icvCvt_BGR2RGB_8u_C3R(unsigned char const*, int, unsigned char*, int, CvSize) () from /usr/local/lib/libopencv_imgcodecs.so.3.3

zivsha, 2017-10-26:
Which version of the library are you using?
There were some bug fixes and behavior changes that were pushed prior to this PR.
Please try the code again with the current development branch.

manhha1402, 2017-10-26:
Hi,
The error showed:
int main(int, char**): Assertion `RS2_FORMAT_BGR8 == color_frame.get_profile().format()' failed.
Aborted (core dumped)

ankit-shinde, 2017-10-26:
Received error
main: color.cpp:37: int main(): Assertion `RS2_FORMAT_BGR8 == color_frame.get_profile().format()' failed.
Aborted (core dumped)

Currently using sdk 2.0, and against master branch

zivsha, 2017-10-26:
@manhha1402 , This is due to a bug that was fixed in PR #734 which is merged already to development.
since we haven't released a version with this fix yet, and since the PR that updates the OpenCV example is still open, you should try to run the code with the latest development.
I've created a branch on my fork for you guys to try this, is has a `opencv` example under the examples folder. You can find it here:
https://github.com/zivsha/librealsense/tree/opencv_with_actual_example

ankit-shinde, 2017-10-26:
Zivsha...
Thanks it worked..

Few questions
1.Can the IR and Depth cam work together?
2.Manual to API for current library?

zivsha, 2017-10-26:
@ankit-shinde 
1. Yes - IR and Depth streams can stream together
2. Please follow our [examples](https://github.com/IntelRealSense/librealsense/tree/development/examples) and you can also follow the Pipeline Changes pull-request (#672) for explanation. Additionally our Wiki holds an [API-How-To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To) that can help

closed 2017-10-26 by ankit-shinde

bradwow, 2017-10-30:
Hi zivsha:
I use your opencv example code 
but it still have problem Assertion `RS2_FORMAT_BGR8 == color_frame.get_profile().format()' failed.
how can i fixed it ,thx~~

zivsha, 2017-10-30:
@bradwow you should pull the latest changes to get the fix for this bug. Just `git pull development` and try again.

bradwow, 2017-10-31:
@zivsha 
thx it can work!
and how can I get depth frame convert to opencv Mat

bradwow, 2017-10-31:
@zivsha 
my project want to use like this demo:
https://github.com/IntelRealSense/librealsense/pull/724/files/35249ed0386d64e4629719dee16c7f7fb1637c01#diff-7e770ad53361509511d2f751db82c939

I want to set threshold to detect different distance object
so I need color and depth frame to opencv and two frame alignment to do others application
how can I align color and depth frame and convert to opencv Mat !?
I'm first time using realsense 
so many question,hope you can help me,thank you!