realsenseai/librealsense #743: Are you going to have skeleton tracking for D435 (SDK 2.0)?

cavisions, 2017-10-26:
| Required Info  |   |
|---|---|
| Camera Model | D400 |

jlgarcia75, 2017-10-26:
Hello @cavisions,

We cannot comment on development roadmaps at the moment but please tell us what your use case and requirements are so that we may scope this feature for the communities needs.

Best regards,
Jesus G.
Intel Customer Support

cavisions, 2017-10-27:
I'm glad to hear from you! So far we've used Microsoft's Kinect sensor to track patient's movements in our software. It is designed to motivate patients to perform movements specified by clinician in a gamified environment. You can have a look at our short [VAST.Rehab video presentation](https://www.youtube.com/watch?v=hO-a7OerEoU) in order to have the basic understanding of what the system is about.

So precisely - what we need is a list of body joints calculated from the depth and RGB data as you can see in [following video](https://www.youtube.com/watch?v=_Ahy0Gh69-M).

It was possible with the RealSense SDK 1.0 to track the upper body this way.

With very kind regards,
Marek

jlgarcia75, 2017-10-27:
@cavisions,

Thank you for your input. I have forwarded your request to the appropriate team. I cannot guarantee that this will be implemented, but it is good to have community feedback like yours for future product development ideas.

Regards,
Jesus G.
Intel Customer Support

closed 2017-10-27 by jlgarcia75

ehtick, 2018-02-23:
Would be great that Intel will go on where Kinect has stopped. But at this point i still get better results with a Kinect then an Intel D435 :(

zivsha, 2018-02-25:
@ehtick , please note that different scenarios require different configurations to provide the best image quality. The D400 series allows users to configure multiple controls and advanced controls of the device, in addition to several predefined presets. You are welcome to read more about it on our Wiki under [D400-Series-Visual-Presets](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets), and you can use the RealSense Viewer to watch the effect on a live stream.

thejae, 2018-03-26:
Now that the D435 has been released, can @jlgarcia75  finally comment on that?

moyayo, 2018-03-26:
Has anyone been able to achieve human pose tracking with the D400 series that is comparable to the Kinect V2 ? (Using a 3rd party middleware like nuitrack)

moyayo, 2018-03-26:
@cavisions have you found a good replacement for the Kinect?

ehtick, 2018-03-27:
I think Intel has a lot off work to do, in comparison to Kinect V2. 
Sorry to say that i am disapointed with the Camera D435 and the SDK :(