realsenseai/librealsense #745: cmake version mismatches

svenevs, 2017-10-26:
It seems the update to `librealsense2` was only partially done.  Right now, in a parent project you have to do something like

```cmake
find_package(realsense2)
if(realsense2_FOUND)
  #                                   v
  list(APPEND THE_LIBS     ${realsense2_LIBRARY})
  list(APPEND THE_INCLUDES ${realsense_INCLUDE_DIR})
  #                                   X
endif()
```

The version numbers also need updating.  The following diff would make sense

```diff
--- a/CMake/realsense2Config.cmake.in
+++ b/CMake/realsense2Config.cmake.in
@@ -1,10 +1,10 @@
 @PACKAGE_INIT@
 
-set(realsense_VERSION_MAJOR "@REALSENSE_VERSION_MAJOR@")
-set(realsense_VERSION_MINOR "@REALSENSE_VERSION_MINOR@")
-set(realsense_VERSION_PATCH "@REALSENSE_VERSION_PATCH@")
+set(realsense2_VERSION_MAJOR "@REALSENSE_VERSION_MAJOR@")
+set(realsense2_VERSION_MINOR "@REALSENSE_VERSION_MINOR@")
+set(realsense2_VERSION_PATCH "@REALSENSE_VERSION_PATCH@")
 
-set_and_check(realsense_INCLUDE_DIR "@PACKAGE_CMAKE_INSTALL_INCLUDEDIR@")
+set_and_check(realsense2_INCLUDE_DIR "@PACKAGE_CMAKE_INSTALL_INCLUDEDIR@")
 
 include("${CMAKE_CURRENT_LIST_DIR}/realsense2Targets.cmake")
 set(realsense2_LIBRARY realsense2::realsense2)
```

however I didn't submit a PR because this could easily break things for people.  Maybe it would be better to, since people are already using these, set both?  I don't think there's a way of deprecating something like this in CMake.

dorodnic, 2017-10-27:
Thanks @svenevs, we will look into this

UnaNancyOwen, 2018-01-15:
I agree with that opinion of @svenevs.
I think that it should change to <code>realsense2_\*</code>.
Do you have a special reason that not adopt this opinion?
Thanks,

ymadzhunkov, 2018-02-01:
Hi, 
if you are going to update this. Please add the following line also

     set(realsense2_VERSION ${realsense2_VERSION_MAJOR}.${realsense2_VERSION_MINOR}.${realsense2_VERSION_PATCH})

So I can write in my project:

     find_package(realsense2 1.9.1 REQUIRED)

svenevs, 2018-02-01:
FWIW, you [can deprecate read access to variables](https://youtu.be/bsXLMQ6WgIk?t=12m41s).  But it's honestly a really weird thing to include in an installed cmake script, and would likely cause more confusion than it does help.

----------------

I think it would be reasonable to just update with the patch above (also including `realsense2_VERSION`), and add a "**Step 3**" in the [quick start section](https://github.com/IntelRealSense/librealsense#quick-start) showing briefly how to `find_package`.  By adding in `realsense2_VERSION`, we could use that as the divide.  The sample code could be something like

```cmake
# find the realsense2 package
find_package(realsense2 REQUIRED)
# assuming you have created a target 'my_target' either
# from add_executable or add_library
target_link_libraries(my_target "${realsense2_LIBRARY}")
# a minor bug was fixed with the introduction of realsense2_VERSION,
# this was introduced in vX.Y.Z [[[ <<< set to actual release ]]]
if (realsense2_VERSION)
  include_directories("${realsense2_INCLUDE_DIR}")
else()
  # minor bug was: missing 2 >>>>v
  include_directories("${realsense_INCLUDE_DIR}")
endif()
```

This snippet shows future users how to account for both pre and post patch with a very simple modification to their own scripts.  Happy to test this if it helps!

@ymadzhunkov do you currently always **fail** or always **succeed** with the version number?  I believe the above approach will also satisfy your requirements, because currently the version is not set so it would fail the version check.  Assuming you are requiring `X.Y.Z` (whenever this is introduced), it should work?

ymadzhunkov, 2018-02-03:
@svenevs Currently it always **fails**.

Regarding your sample code, since cmake v3.0

     target_link_libraries(my_target "${realsense2_LIBRARY}")

will also automatically add public include path of ${realsense2_LIBRARY} to my_target. Therefore, you don't need to write include_directories. If you want I can prepare a Pull Request that resolves this issue.

svenevs, 2018-02-05:
> Currently it always fails.

Ok cool, so this should meet all criteria.

> Therefore, you don't need to write `include_directories`

Very interesting, I didn't know about this!  I think it's growing less and less likely that people are using <3.0 for CMake, but 2.8.12 still seems to be a default.  So maybe include the small blurb first just up to `targe_link_libraries`, and then say "If, however, you are using CMake less than 3.0, you also need to ... `include_directories` example"?

Part of the reason I never submitted a PR is because the maintainers here need to make a decision on how they want to approach it.  The options I see are

1. Set both, allowing for existing projects that already have it using the "missing 2" version to still function.
2. Set the correct one, adding docs explaining how to support both versions.
3. Fix it, and ignore.

I think (2) is the right approach, but doing either (2) or (3) pushes the problem downstream to any projects currently using `librealsense`.  So at some future point, some project `foo` will have an issue raised saying "I can't compile with `librealsense`", having docs on the main README will hopefully help people solve it faster.

The primary argument for not doing `1` is if anybody rolled out their own `FindLibRealSense.cmake` for **version 1.x**, setting both will potentially create conflicts for them.

> If you want I can prepare a Pull Request that resolves this issue.

I think that would be great, I'm sure the maintainers would appreciate the head-start instead of trolling through this issue :)

svenevs, 2018-03-01:
@jlgarcia75 is there a reason this was closed?  The problem persists [on the development branch](https://github.com/IntelRealSense/librealsense/blob/development/CMake/realsense2Config.cmake.in).

dorodnic, 2018-11-21:
Should be finally addressed in 2.16.5 (see the [release notes](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features--improvements))
@svenevs @UnaNancyOwen thank you for the feedback. 
You are welcomed to review the updated [CMake files](https://github.com/IntelRealSense/librealsense/blob/master/CMakeLists.txt) and comment on any additional problems

closed 2018-11-27 by RealSense-Customer-Engineering