realsenseai/librealsense #749: Support for C# wrapper

jernejk, 2017-10-28:
| Required Info  |   |
|---|---|
| Camera Model | SR300 / D400 | 
| Operating System & Version | Windows 10 |

Are there any plans for supporting C# and perhaps UWP as well?

dorodnic, 2017-10-29:
C# wrapper - certainly.
UWP is a bit more tricky - at the moment, `librealsense` is using `Windows Media Foundation` and `WinUSB` APIs to communicate with the device, both need to be changed (to `Media Capture` API instead) There is no clear timeline for that at the moment but it can change soon.

NikoBele1, 2017-11-20:
Hey @dorodnic do you know when the C# wrapper will be availlable?

dorodnic, 2017-11-22:
It is under development together with additional wrappers, but I cannot provide a specific date. As a general trend, we try to include one new wrapper with every new release we make. This week it was LabVIEW. C# and Matlab are currently the two under development and will be published once they meet sufficient quality.

dorodnic, 2017-12-17:
Initial version is now under review (#917).
You are welcomed to try it out and comment. We will be happy to incorporate feedback.

closed 2017-12-24 by dorodnic