realsenseai/librealsense #750: Handouts for examples' cmake

wwangzhao, 2017-10-28:
| Required Info  |   |
|---|---|
| Camera Model |  SR300  | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.10.0-37-generic  |

Hi 

I am a new user of Ubuntu and just start a new project with SR300. Previously I have many experience on Matlab so I can read and understand the code. 

Would you pls give a detail description of how to use cmake to compile the examples code?

For instance, I am trying to work with the example of capture
 I have tried 

mkdir build && cd build
cmake ../ 
make

Then, I don't know how to continue, How can I get the executable?
Best Regards
Joe

zivsha, 2017-10-29:
Hi @wwangzhao , Please try to follow the [Linux Installation guide](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md)

dorodnic, 2017-10-29:
You are in the right direction, please try:
```
mkdir build && cd build
cmake .. -DBUILD_EXAMPLES=true
make
sudo make install
rs-capture
```
Let us know if this helps

wwangzhao, 2017-10-30:
Hi, @zivsha  and @dorodnic 

Thx for your reply. It seems that I haven't make it clear.
For the linux installation guide, I have finished the installation and the examples works well, e.g. the rs-capture, rs-align, etc. 
However, I wish to start my own project. Hence I try to compile the code in the example, e.g.  in the example of align, I tried

mkdir build && cd build
cmake ../ -DBUILD_GRAPHICAL_EXAMPLES=true
make

Then it shows
Scanning dependencies of target rs-align
[ 20%] Building CXX object CMakeFiles/rs-align.dir/rs-align.cpp.o
[ 40%] Building CXX object CMakeFiles/rs-align.dir/home/james/librealsense/third-party/imgui/imgui.cpp.o
[ 60%] Building CXX object CMakeFiles/rs-align.dir/home/james/librealsense/third-party/imgui/imgui_draw.cpp.o
[ 80%] Building CXX object CMakeFiles/rs-align.dir/home/james/librealsense/third-party/imgui/imgui_impl_glfw.cpp.o
[100%] Linking CXX executable rs-align
CMakeFiles/rs-align.dir/rs-align.cpp.o: In function `rs2::error::error(rs2_error*)':
rs-align.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x2a): undefined reference to `rs2_get_error_message'

There are a lot of messages about undefined reference to `rs2_get_error_message', it seems that the rs.h is not included? I find that the forlder _librealsense2_ is under /usr/local/include
What should I do then ?

Best Regards

zivsha, 2017-10-31:
Hi @wwangzhao,
I've created a simple "getting started" example on my fork which you can take and build as a stand alone project: [zivsha/getting-started](https://github.com/zivsha/librealsense/tree/getting_started_example/examples/).
It is not the best way to create a `cmake` project but it is something to help you get started.

Note that you must install the library prior to building this project. Also this will probably not work on windows unless you add the path to the realsense.dll to your environment variable.
We will consider adding such a template project to the SDK itself, with a more robust cmake, and detailed readme in the near future.

wwangzhao, 2017-10-31:
Hi @zivsha 

Thanks a lot for your reply. I have tried the rs-getting-started. The CMake and Make goes well but it reports error for ./rs-getting-started:

RealSense error calling rs2_get_frame_width(frame_ref:nullptr):
    null pointer passed for argument "frame_ref"

I noticed that the light (both green and red) turns on once, and I also tried the executable example of rs-capture, which shows that the camera works well. 

Anyway, when I use ldd rs-getting-started, it shows the library links well:
$ ldd rs-getting-started 
	linux-vdso.so.1 =>  (0x00007fffe3b99000)
	librealsense2.so.2 => /usr/local/lib/librealsense2.so.2 (0x00007ff6cc36d000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ff6cbfa0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ff6cbd88000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff6cb9be000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007ff6cb7a6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff6cb49c000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007ff6cb26c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff6cb04f000)
	/lib64/ld-linux-x86-64.so.2 (0x000055db84ca8000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007ff6cb02e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007ff6cae2a000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007ff6cac22000)

zivsha, 2017-11-01:
Hi @wwangzhao ,
This is caused because the `frames.get_depth_frame();` returned an empty frame. 
This should not happen currently (in `development`) but it is not yet merged to `master`.
All you have to do is add:
```cpp
        // Try to get a frame of a depth image
        rs2::depth_frame depth = frames.get_depth_frame();
        if(!depth) continue;  // <--------------------------- Add this
        // Get the depth frame's dimensions
```

zivsha, 2017-11-02:
Closing the issue, as it seems like you've managed to get started.

closed 2017-11-02 by zivsha