realsenseai/librealsense #754: rs2_pipeline_start_with_config Device or resource busy

dlouisroberts, 2017-10-30:
| Required Info  |   |
|---|---|
| Camera Model | 2x SR300  | 
| Firmware Version |  3.15.0.0 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.10.0 |

I am working on an application with realsense devices. Currently I'm using two SR300s, but expect to have one R200 and one SR300 soon. 

In making some modifications to rs-multicam.cpp, I have encountered an issue where starting the pipeline after calls to rs2::config.enable_stream() results in the following error:

`
realsense error calling rs2_pipeline_start_with_config(pipe:0x19a6010, config:0x19ba420):
    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
`
Here are the lines of code where I'm making those calls:
`
		rs2::config c;
		c.enable_device(serial_number);
		c.enable_stream( RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30 ); // <-- modified line
		c.enable_stream( RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30 ); // <-- modified line
`
The line that triggers the error is:
`		rs2::pipeline_profile profile = p.start(c);`

While debugging, I have determined that this error only occurs when starting the pipeline for the second device---it works correctly when only a single device is connected (at least until calls to texture.upload(), but that's a separate issue I can fix). 

The example runs correctly without modifications, indicating the cameras are performing correctly with the default configurations. I have also separately verified with rs-realsense-viewer that I can stream both cameras with the desired configuration correctly. 

Thanks in advance!

-Dave

zivsha, 2017-10-30:
Hi @dlouisroberts,
The error looks like it is caused due to opening an already streaming camera.
1. Which version of the SDK are you using?
2. Can you please verify that the calls to `c.enable_device(serial_number);` contain different values?

Jason420Git, 2017-10-30:
Hi,

I have also experienced this issue with multi-cam streaming. It seems to be when you try to configure more than one device. It is fine if you do not set the stream configuration. But if you do, (even with setting different serials) it seems after .start() the pipe becomes the same device.

zivsha, 2017-10-30:
We had a bug with `pipeline::start` which could have caused this. Can you try to reproduce this with the current `development`?

dlouisroberts, 2017-10-30:
I upgraded to the development branch and the issues is resolved. Thanks!

zivsha, 2017-10-31:
@Jason420Git were you also able to resolve it?

closed 2017-11-01 by dorodnic

ywcmaike, 2018-02-01:
I also encountered the questions like that. I did just like the document how to build  from the source code. https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md 
It runs normally. When I want to capture the picture,the window has flashed back。 and it runs just like this as followed:
/usr/local/bin$ ./rs-capture 
RealSense error calling rs2_pipeline_start(pipe:0x1f91b60):
    libusb_claim_interface(...) returned LIBUSB_ERROR_BUSY Last Error: Device or resource busy

How to resolove it ?

My realsense is D415. And my system is Ubuntu16.04.

ywcmaike, 2018-02-01:
And I also reboot, but it make no sense.

zivsha, 2018-02-01:
Maybe, but when you hold a `rs2::frame` you can call:
```cpp
frame.get_profile().stream_name()
```
for a human readable string.