realsenseai/librealsense #757: How to convert pointcloud to PCL point cloud

manhha1402, 2017-10-30:
| Required Info  |   |
|---|---|
| Camera Model |SR300
| Firmware Version |   | 
| Operating System & Version |  14.04 |
Hi,
I am trying to convert stream of SR300 camera to point cloud in PCL library. I see in the example "rs-pointcloud.cpp", we can generate point cloud from depth and color stream. But the example uses OpenGL to visualize the point cloud. 
Anyone knows how to convert streams to pcl::PointCloud<pcl::PointXYZRGB> in PCL library ?
Thanks

manhha1402, 2017-11-03:
I can do it. Closed!!

closed 2017-11-03 by manhha1402

Jiloc, 2017-11-03:
hello @manhha1402, could you post an example? I am also interested

jjoachim, 2017-12-28:
Bump? Interested for a D415 camera. I need a pointcloud (PointRGBXYZ) with 3d points in meters and registered color data. I've only had partial luck so far with tweaking the current examples...