realsenseai/librealsense #761: Depth settings causing error

Jason420Git, 2017-10-31:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version |  5.8.5 | 
| Operating System & Version |  Windows 10 |

I have been unable to set the laser emitter power or depth presets without getting the following error:

"RealSense error calling rs2_query_sensors(device:nullptr):
null poiinter passed for argument "device"

I am setting it with the following line of code:

```
m_rsDevice.first<rs2::depth_sensor>.set_option(rs2_option::RS2_OPTION_LASER_POWER, val);
```

I have tried this both before and after starting the camera

zivsha, 2017-10-31:
Hi @Jason420Git , 
The error you are getting indicates that your `m_rsDevice` is "empty", meaning that you did not initialize it with a valid device.
Please verify that it is valid (with `if(m_rsDevice) { /*use m_rsDevice*/ }`, and check you code before this validation to make sure you set it with a valid device.

Jason420Git, 2017-10-31:
I was able to get this working. My setup may be a bit odd as I have my rs_device as a member to a class, I pass in the device from a function. It is a wrapper class around the realsense API.

closed 2017-11-01 by dorodnic