realsenseai/librealsense #764: OpenCV with Python wrapper example

alered87, 2017-10-31:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 616205004967 3.21.0.0  | 
| Operating System & Version | Ubuntu 10.4  |
| Kernel Version (Linux Only)  |  4.10.0 |

Hi, I encountered the following error when running "opencv_viewer_example.py":

Traceback (most recent call last):
  File "opencv_viewer_example.py", line 32, in <module>
    images = np.hstack((color_image, depth_colormap))
  File "/home/udoo/.local/lib/python2.7/site-packages/numpy/core/shape_base.py", line 293, in hstack
    return _nx.concatenate(arrs, 1)
ValueError: all the input arrays must have same number of dimensions

which seems to be due to the fact that 'config.enable_stream()' reshape the color image to a 2 dimension array, dropping the channels dimension and converting data type to Y16 (printed from get_profile())

I have problem to the profile of color image when both color and depth are initialized

Thank you

zivsha, 2017-11-02:
Hi @alered87 ,

Can you please try to rebuild the python binding again with the latest `master` or `development` and try to run the sample again?
This looks like the same issue as #701

zivsha, 2017-11-06:
@alered87 , please reopen if the issue is different/not resolved with new version

closed 2017-11-06 by zivsha

alered87, 2017-11-09:
Hi @zivsha ,

the new version works, thank you!

sorry for the late feedback

zivsha, 2017-11-09:
Good to know 😃