realsenseai/librealsense #765: No camera control error

Jason420Git, 2017-10-31:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | 5.8.5  | 
| Operating System & Version |  Windows |

I am trying to get my streaming to be robust and an error I sometimes encounter is the following:

[image]

I m using the device to create HDR images so I am changing the exposure often. Is there some kind of lockup occuring that I can try to avoid? Many thanks.

dorodnic, 2017-10-31:
Hi @Jason420Git 
Any chance you can attach the full call-stack of the exception?

Jason420Git, 2017-10-31:
[image]

It happens a lot less frequently in Debug mode but I eventually got it to crash. I would be running in Release.

dorodnic, 2017-10-31:
We have not previously encountered this error. If you could share the code snippet you are using and the exact Windows build number, we will try to reproduce it and provide a fix. There are significant changes in Windows Media Foundation behavior between different builds of Windows 10.

Jason420Git, 2017-11-01:
It may also be a problem with how I've set up my data. I have a RealSense class which is my own wrapper around the RealSense API. Since I use multiple cameras I have a Camera class which describes the whole system. Here is a stripped down version of the main components. Note, that I can change the colour exposure successfully but at times it does not work. 

```
class RealSense
{
   private:

   rs2::pipeline m_rsPipe;
   rs2::device m_rsDevice;
   rs2::config m_rsConfig;
    rs2::pipeline_profile m_rsPipelineProfile;

public:
   RealSense(rs2::device dev, std::string serial)
   {
      m_rsConfig.enable_device(serial);
   }

  bool start()
  {
    	m_rsPipelineProfile = m_rsPipe.start(m_rsConfig);
  }

  bool setColourExposure(float val)
  {
	if (m_rsDevice.query_sensors()[1].supports(rs2_option::RS2_OPTION_EXPOSURE))
	{
		setColourAutoExposure(false);
		m_rsDevice.query_sensors()[1].set_option(rs2_option::RS2_OPTION_EXPOSURE, val);
		return true;
	}

	return false;
  }
};

class Camera
{
   rs2::context ctx;
   rs2::device_list deviceList;
   std::vector<RealSense> realsenseList;
   std::vector<std::string> serialList;
  
public:
   Camera()
   {
      serialList.push_back("7710....") // Device serial here
      serialList.push_back("7310....") // Device serial here
   }
   

  void addDevices()
  {
      deviceList = ctx.query_devices();

      	for (auto&& dev : deviceList)
       {
          std::string currentSerial = dev.get_info(rs2_camera_info::RS2_CAMERA_INFO_SERIAL_NUMBER);
	 
          if (currentSerial == deviceSerials[s])
           {
             Realsense rsDev(dev, currentSerial);
             realsenseList.push_back(dev);
           }
       }
  }

   //Change the exposure on all of the realsenses (run separate threads)
   void changeExposures()
   {
        float exposureToSet = -5;
        std::vector<boost::thread *> z;
       	for (int c = 0; c < camList.size(); c++)
        {
            z.push_back(new boost::thread(&Camera3D::exposureChangeThread, this, &realsenseList[c], exposureToSet ));
        }

	for (int i = 0; i < z.size(); ++i)
	{
		z[i]->join();
		delete z[i];
	}
   }

void exposureChangeThread(RealSense *dev, float expoVal)
{
      dev->setColourExposure(expoVal);

     //Grab 15 frames to settle
     ...

}
};

```

Jason420Git, 2017-11-01:
Windows 10 Home.

Build 15063.674

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
@Jason420Git
Is the problem still there?

closed 2018-05-28 by RealSense-Customer-Engineering