realsenseai/librealsense #766: SR300 is not detected in realsense-viewer

hironoriyh, 2017-11-01:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04 LTS |
| Kernel Version (Linux Only)  |  4.10.0-38-generic |
| Hardware | Surface Book performance base |

I followed [this link](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md), but realsense-viewer says please connect device. 

The result from 
`modinfo uvcvideo | grep "version:" `
is 
`version:        1.1.1.realsense-4.4.0
srcversion:     F394CA611A53D8EEC317046
`

and  `sudo dmesg | tail -n 50
 `
is 
`[  664.005629] usb 2-1.1.1.1: New USB device found, idVendor=8086, idProduct=0aa5
[  664.005633] usb 2-1.1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  664.005636] usb 2-1.1.1.1: Product: Intel(R) RealSense(TM) Camera SR300
[  664.005639] usb 2-1.1.1.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[  664.005641] usb 2-1.1.1.1: SerialNumber: 61020100408` 

also lsusb shows
`Bus 001 Device 010: ID 041e:4099 Creative Technology, Ltd 
`

dorodnic, 2017-11-01:
Do you see the camera using `lsusb`?

hironoriyh, 2017-11-01:
@dorodnic  with lsusb, yes, Creatie Technology is SR300 camera. The other laptop, which is working with librealsense v1, returns the same. So at least camera is recognized.

ev-mp, 2017-11-01:
Hi @hironoriyh , what is the output for "rs-enumerate devices" demo (from librealsense2-utils package)?
Additionally, please log disconnect/reconnect flow and provide the full log with "dmesg -wT"

ev-mp, 2017-11-09:
@hironoriyh , do you still need assistance with the topic ?

hironoriyh, 2017-11-09:
@dorodnic  sry for the late response. we use sr300 on another machine now.

ev-mp, 2017-11-13:
Hi @hironoriyh , I'm closing this it seem you have the device running properly on another PC.
Feel free to reopen this in the future.

closed 2017-11-13 by ev-mp