realsenseai/librealsense #769: rs2_get_depth_scale

bradwow, 2017-11-01:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 |  SR300
| Firmware Version |   | 
| Operating System & Version |   | WINDOWS 10
| Kernel Version (Linux Only)  |   |

how can i use this function : rs2_get_depth_scale
i need to translate depth data to (mm)

and how can i convert depth to color frame to opencv mat type

bradwow, 2017-11-01:
and i have same question here
https://github.com/IntelRealSense/librealsense/issues/742

zivsha, 2017-11-01:
Hi @bradwow ,

I noticed that your question in #742 indicates that you went over the `rs-align` example.
First, I would appreciate it if you could tell me which parts were unclear in the example itself so we may improve it.
Second, as you can see in the examples code in order to get the depth scale you need to get the depth sensor from the device, and then using the depth sensor call `get_depth_scale()` (In `C` it would be the same flow of getting the sensor, and then calling `rs2_get_depth_scale(depth_sensor, &e)`)
https://github.com/IntelRealSense/librealsense/blob/82e04ee1a34fda79a9ea5bbf0366601c28fbf9f9/examples/align/rs-align.cpp#L128-L140
Another way is using this one-liner: 
`float depth_scale = profile.get_device().first<rs2::depth_sensor>().get_depth_scale();` 
> Note that this line could throw exceptions from various functions, and there is no null validations!

Additionally, the example shows how to align depth frames to other frames, which is done using the `rs2::align` object:
 ```cpp
rs2::align align(RS2_STREAM_COLOR);
while(/*getting frames*/)
{
        rs2::frameset frameset = pipe.wait_for_frames();
        rs2::frameset proccessed = align.proccess(frameset);
        rs2::video_frame color_frame = proccessed.first(RS2_STREAM_COLOR);
        rs2::depth_frame aligned_depth_frame = proccessed.first(RS2_STREAM_DEPTH);
}
```

putuyuwono, 2017-11-02:
Hi, 
I hope this might help you out

**Convert Color Frame to Mat**
```
Mat colorMat(Size(colorFrame.get_width(), colorFrame.get_height()), CV_8UC3,
 (void*)colorFrame.get_data());
cvtColor(colorMat, colorMat, COLOR_RGB2BGR); //dont forget that imshow expect BGR image
imshow("Color Mat", colorMat);
```

**Convert Depth Frame to Mat**
```
Mat depthMat(Size(depthFrame.get_width(), depthFrame.get_height()), CV_16UC1,
 (void*)depthFrame.get_data());
```

closed 2017-11-06 by zivsha

bradwow, 2017-11-10:
thank you ! i will try it!

jinfagang, 2018-12-17:
@zivsha  Hi, I am trying this but got error:

```

    rs2::device device;
    rs2::sensor& sensor = device.query_sensors().front();
    // in meters, pixel value 1 * rs_real_distance_scale
    rs_real_distance_scale = sensor.as<rs2::depth_sensor>().get_depth_scale();
    cout << "Find realsense camera depth real distance scale: " << rs_real_distance_scale << endl;
```
Erro:

```
terminate called after throwing an instance of 'rs2::error'
  what():  null pointer passed for argument "device"

```

Any suggestions?

zivsha, 2018-12-18:
Your device's underlying handle is null. You can't just create a device. You need to get the device from a `context` object, or from an active pipeline profile. Look at the example page to see how.