realsenseai/librealsense #770: How can I change the rate of IMU in realsense 300?

neil-qi, 2017-11-01:
How can I change the rate of IMU in realsense ZR300?
When I tested, imu and fisheye's time difference is 20ms ,but it is not good  enough. What can I do?
ps:  fisheye_strobe  and fisheye_external_trigger  is set to 1.
Thank you.

dorodnic, 2017-11-01:
The IMU rate in the ZR300 peripherals is fixed (I believe it was 250 FPS for the accelerometer and 200 for gyro) This means that when using callbacks you can expect to receive motion data every 4-5 ms.

neil-qi, 2017-11-02:
Thank you for your reply.
when I use 
$rostopic hz /camera/imu/data_raw
subscribed to [/camera/imu/data_raw]

average rate: 442.345
	min: 0.000s max: 0.013s std dev: 0.00162s window: 388
average rate: 437.868
	min: 0.000s max: 0.013s std dev: 0.00153s window: 822
average rate: 439.314
	min: 0.000s max: 0.013s std dev: 0.00152s window: 1264
average rate: 438.738
	min: 0.000s max: 0.013s std dev: 0.00152s window: 1702
average rate: 438.145
	min: 0.000s max: 0.013s std dev: 0.00154s window: 2138
average rate: 436.783
	min: 0.000s max: 0.013s std dev: 0.00152s window: 2568
average rate: 436.837

dorodnic, 2018-01-13:
I believe this is correct

closed 2018-01-13 by dorodnic