realsenseai/librealsense #775: Python - Get Depth aligned to RGB

dolhasz, 2017-11-02:
| Required Info  |   |
|---|---|
| Camera Model |SR300 | 
| Firmware Version |  2 | 
| Operating System & Version |  Windows 10 |
| Kernel Version (Linux Only)  |   |

Hi there,

Does the Python wrapper support mapping the depth map to the RGB stream? I can't seem to get it working.

Thanks,

Alan

zivsha, 2017-11-02:
Hi @dolhasz ,

It seems that we've missed the binding for this object. Thanks for letting us know.
I will add it soon!

zivsha, 2017-11-02:
Hi again :)

I've opened PR #776  to use the alignment processing block of the SDK.
Also, I've composed a small example which tries to imitate the `rs-align` example (i.e background removal).
> NOTE: I'm not an expert in `numpy` and I'm probably masking the color image incorrectly

```py
import pyrealsense2 as rs
import numpy as np
import cv2
# Create a pipeline
pipeline = rs.pipeline()
#Create a config and configure the pipeline to stream
#  different resolutions of color and depth streams
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
# Start streaming
profile = pipeline.start(config)
# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print "depth_scale : " , depth_scale
# Create an align object
align = rs.align(rs.stream.color)
# Streaming loop
try:
    while True:
        # Get frameset of color and depth
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        print "Depth Resolution: " , depth_frame.get_width() , "x" , depth_frame.get_height()
        # Align the depth frame to color frame
        proccessed = align.proccess(frames)
        # Get aligned frames
        aligned_depth_frame = proccessed.get_depth_frame()
        print "Aligned Depth Resolution: " , aligned_depth_frame.get_width() , "x" , aligned_depth_frame.get_height()
        color_frame = proccessed.get_color_frame()
        if not aligned_depth_frame or not color_frame:
            continue
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        # My lousy attempt to perform background removal
        depth_image_3d = np.repeat(depth_image[:, :, np.newaxis], 3, axis=2)
        depth_image_3d = depth_image_3d * depth_scale
        bg_removed = np.where(depth_image_3d > 1, 0, color_image)
        # Render images
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, None, 0.5, 0), cv2.COLORMAP_JET)
        images = np.hstack((bg_removed, color_image, depth_colormap))
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)
finally:
    pipeline.stop()
```

zivsha, 2017-11-06:
Closing since PR #776 was merged.

closed 2017-11-06 by zivsha