realsenseai/librealsense #777: repo for ubuntu 17.10

titoasty, 2017-11-03:
Hi,

I have a fresh install of Ubuntu 17.10 and when adding the realsense repo, I've got a 403 on apt-get update...
Here are the tasks I've followed, according to doc :
```
echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 7DCC6F0F
sudo apt-get update
```
And I've got a 403 forbidden on http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo
Thanks in advance for your help !

The-Funk, 2017-11-27:
Hello! Same exact issue here. I get a 403 when using https as well.

aloisklink, 2017-12-05:
Hi, I'm having the same issue, I'm getting the error:

`E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  403  Forbidden`

https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease seems to return a 403 error, while the following all work:

- https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/Release
- https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/main/binary-amd64/Packages.gz
- https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/main/binary-amd64/Packages
- https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/main/binary-amd64/Release

Normally `apt` looks for an `InRelease` file, and if it gets a HTTP 404 file not found error, it should move to looking for a `Release` file. So @dorodnic either you need to change your server so it returns a HTTP 404 when looking for the `InRelease` file, instead of a HTTP 403 (which may be difficult depending on who controls your server), or create an `InRelease` file.

For all of you guys that still want to install this, try entering the following commands into your terminal, it manually downloads the files that apt requires, basically manually doing `sudo apt update` so you can do `sudo apt-get install librealsense2-utils`.
```bash
cd /var/lib/apt/lists
sudo wget http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/main/binary-amd64/Packages -O realsense-hw-public.s3.amazonaws.com_Debian_apt-repo_dists_xenial_main_binary-amd64_Packages
sudo wget http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/Release -O realsense-hw-public.s3.amazonaws.com_Debian_apt-repo_dists_xenial_Release
```

dorodnic, 2017-12-10:
Hi,
Wanted to update that we are looking into this problem. 
@aloisklink thank you very much for your comment, I'll try to modify the permissions as you suggested. 
For now, the alternative is building the library from source ([instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)).

Jiloc, 2017-12-28:
same issue here, I had to follow @aloisklink instructions

peteflorence, 2018-01-26:
Similar issue here:

E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/Release  Unable to find expected entry 'main/source/Sources' in Release file (Wrong sources.list entry or malformed file)

bedge, 2018-02-16:
Is this because with s3 permissions when you request a file that doesn't exists and you don't have read-access to the bucket, you get a 403 rather than the expected 404 for the file actually not being there?
If this is the case, I expect one could solve it be opening up the bucket permissions to allow directory reading. 
Can one create a file with some permission set on s3 that would force a 404? Perhaps if InRelease is a directory rather than a file? I'm guessing not.

dorodnic, 2018-04-03:
We don't plan to add Ubuntu 17 PPA in the upcoming Sprints, but will keep it in the backlog and revisit once more people start using it.

closed 2018-04-03 by dorodnic

The-Funk, 2018-04-03:
I think most people are just wondering why the Ubuntu 16 repository returns a 403 forbidden when trying to to download the SDK from a Ubuntu 17 machine.

Shouldn't the 16 repository work on 17 (despite a few possible dependency issues?) They're both apt.

Do you plan to support Ubuntu 18? The only thing keeping me from using the SDK is the lack of package management. I'd rather not install from source if I don't have to.

aloisklink, 2018-04-14:
@The-Funk, in Ubuntu 16.04, when apt gets **any** HTTP error code for the `InRelease` file, apt would then look for a Release file. However, apt was updated (see https://github.com/Debian/apt/commit/cbbf185c3c55effe47f218a07e7b1f324973a8a6 for the specific commit) so that it would only happen for HTTP 404 errors. So even though they are both apt, one is newer than the other :smile:.

It might be worth making a commit for apt that fixes this, because I can imagine some other repos might still not be using `InRelease` files and are configured to give 403 errors instead of 404 Errors for security reasons (like this one). 

@dorodnic, it might just be easier to make an `InRelease` file for the repo, (see [here](https://help.ubuntu.com/community/CreateAuthenticatedRepository) for the guide), especially since most Ubuntu 16.04 LTS users would start switching over to 18.04 LTS at the end of this month, which has an updated version of `apt`.

You should just need to run one command, `gpg --clearsign -o InRelease Release` and that should make the `InRelease` file (assuming the instructions in the above link are accurate). And that should fix it for everyone that has an apt that's been updated since 2017.

r4space, 2018-08-31:
I know this is closed but I cannot find a solution - what is the current recommended means of installing  intel-realsense-dfu*  on ubuntu 16 or 18?

aloisklink, 2018-08-31:
<https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md> seems to suggest that the repo has been updated for Ubuntu 18.04 bionic.

If that doesn't have what you want, you'd probably need to build everything manually, which might take some time, but there are some pretty good instructions for Ubuntu so you just need to copy and paste bash commands: <https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md>