realsenseai/librealsense #779: No camera detected with patches applied

Maxime59, 2017-11-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 & ZR300 |
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04.03 LTS  | 
| Kernel Version (Linux Only)  | 4.4.0-98-generic  | 

Hi everyone,

I bought the elements of a development kit like this one : https://up-shop.org/home/117-up-zr300-realsense-camera-robotic-developement-kit.html
The only difference is that my UP board has 2GB RAM & 16 Gb eMMC.

I've installed Ubuntu following those steps : https://01.org/developerjourney/recipe/intel-realsense-robotic-development-kit#introduction
I'm trying to setup everything to work with the SR300 or ZR300 on the USB 3.0 OTG port available on my board as the steps should be the same for setting up the system even if the mentionned camera is a R200 in this tutorial.
I've tried to install the upboard headers as mentionned but the kernel patch is not applied fully if these headers are installed instead of the generic ones.

After following those steps I've followed the ones on this github installation page for the ZR300 : https://github.com/IntelRealSense/librealsense/blob/v1.12.1/doc/installation.md
And the steps described in the new version of the SDK for the SR300 : https://github.com/IntelRealSense/librealsense

 I successfully applied the patch to the generic headers and the device seems to be detected as the `dmesg` command shows : 
* For SR300 :
```
[ 5680.371803] usb 1-1: new high-speed USB device number 18 using xhci_hcd
[ 5680.501708] usb 1-1: New USB device found, idVendor=05e3, idProduct=0610
[ 5680.501734] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 5680.501753] usb 1-1: Product: USB2.0 Hub
[ 5680.501768] usb 1-1: Manufacturer: GenesysLogic
[ 5680.503339] hub 1-1:1.0: USB hub found
[ 5680.503581] hub 1-1:1.0: 2 ports detected
[ 5681.639908] usb 1-1.2: new full-speed USB device number 19 using xhci_hcd
[ 5681.816728] usb 1-1.2: config 1 has an invalid interface number: 3 but max is 2
[ 5681.816754] usb 1-1.2: config 1 has no interface number 2
[ 5681.886980] usb 1-1.2: New USB device found, idVendor=041e, idProduct=409b
[ 5681.886999] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5681.887017] usb 1-1.2: Product: BlasterX Senz3D VF0810
[ 5681.887033] usb 1-1.2: Manufacturer: Creative Technology Ltd
[ 5681.887046] usb 1-1.2: SerialNumber: 170224185111
[ 5682.035461] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1.2/1-1.2:1.3/0003:041E:409B.0006/input/input13
[ 5682.089425] hid-generic 0003:041E:409B.0006: input,hiddev0,hidraw3: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:14.0-1.2/input3
```
* For ZR300 :
```
[ 8248.066263] usb 1-1: new high-speed USB device number 20 using xhci_hcd
[ 8248.196174] usb 1-1: New USB device found, idVendor=0451, idProduct=8027
[ 8248.196190] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[ 8248.196195] usb 1-1: SerialNumber: F00F0089A8EF
[ 8248.197812] hub 1-1:1.0: USB hub found
[ 8248.197916] hub 1-1:1.0: 2 ports detected
[ 8248.470246] usb 1-1.2: new high-speed USB device number 21 using xhci_hcd
[ 8248.559423] usb 1-1.2: New USB device found, idVendor=04b4, idProduct=22c3
[ 8248.559433] usb 1-1.2: New USB device strings: Mfr=1, Product=6, SerialNumber=0
[ 8248.559437] usb 1-1.2: Product: CX3-WinUsb
[ 8248.559441] usb 1-1.2: Manufacturer: Cypress
```
The devices appear as Creative or Cypress devices but not as Intel ones and in fact only the built-in microphone is detected for the case of the SR300. It remains impossible even to simply access the RGB stream on both devices through cheese, guvcview or realsense-viewer.

I tried to enabled the ZR300 detection parameter that is present into the BIOS of the upboard but without any success.

Any help on such issue? 
Have I done something wrong during the setup? 
Have you ever had some similar issues with an UP board or other platforms?

Thank you in advance for your help!

Max

freemanlo, 2017-11-07:
It looks your USB related log for RealSense device is weird. Please remember USB3 port is required, instead of USB2 port.
For ZR300 peripheral: 
use "lsusb" command, you should see two USB devices found.
1) VID:PID = 8086:0ad0: this is "CX3-UVC" (Cypress)
2) VID:PID = 8086:0acb, this is "Intel RealSense 3D Camera ZR300"

For SR300:
use "lsusb" command, you should see the SR300 USB device found.
VID:PID = 8086:0aa5: this is "Intel RealSense Camera SR300"

jlgarcia75, 2017-11-09:
Hello @Maxime59,

You should be aware that the minimum processor for the SR300 camera is a 6th Gen Intel Core or later. Therefore, it is not compatible with the Up Board. 

As @freemanlo states, make sure you are connecting the ZR300 to the micro-B superspeed port next to the HDMI port on the Up Board.

Regards,
Jesus G.
Intel Customer Support

Maxime59, 2017-11-09:
Hello @jlgarcia75 

Actually I managed to make the SR300 work with the UP board.
The problem was coming from the USB 3.0 OTG adapter I was using.

EdXian, 2017-11-12:
Hi @Maxime59 ,
Can your UP board work with SR300 or R200?
I also follow the install guide [here](https://01.org/developerjourney/recipe/intel-realsense-robotic-development-kit#introduction) but still cant detect my camera through command `lsusb`.
I installed the `up-board` kernel first and I found that my kernel was substitue with `linux generic` after install [intel package](http://wiki.ros.org/librealsense#Installation_Prerequisites).
Any idea?

Maxime59, 2017-11-13:
In my case I didn't use the up-board kernel in order to be able to install the patch provided that are made for generic distributions. 
Install back the generic headers and then delete the up-board ones using this command : 
`sudo apt -y autoremove --purge 'linux-.*up-board'`
In the end the key for me was to use this adapter : https://www.amazon.fr/gp/product/B00SN1BOW6/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

EdXian, 2017-11-13:
Hi @Maxime59 ,
I think my problem is due to my USB3.0 adapter which is not support OTG . 
I will try other USB3.0 cables tommorow.
Thank you very much.

closed 2017-11-27 by jlgarcia75