realsenseai/librealsense #780: RGB <-> Depth correspondence

yongduek, 2017-11-06:
| Required Info  |   |
|---|---|
| Camera Model | R200 /  SR300 | 
| Operating System & Version |   Ubuntu 16.04, Mac OS X 10.12|

I have been looking for two mandatory functions to do the following:
1. Given the rgb pixel coordinate (i,j), how to find its corresponding depth image coordinate (u,v) ?
2. Conversely, what is the rgb pixel coordinate (i,j) corresponding to the depth image coordinate (u,v) ?

For example, when an object in RGB is detected. How can I get the point cloud data { (x,y,z) } corresponding to the object pixels so that I can draw it in 3D with OpenGL?

It seems that real sense is sold with its calibration parameters, so I think there must be functions to do this. However, even though I have been looking for an answer quite long time, I could not find it.  Hopefully, someone may have a simple answer.

Many thanks in advance.

danielw256, 2017-11-27:
I will need to do this too. Someone please answer.

Link below suggest rs-align has some code to do it, but it has problems too.
https://github.com/IntelRealSense/librealsense/issues/837

jlgarcia75, 2017-11-27:
Hello @danielw256 @yongduek,

In the SDK 2.0, the current master version, projection functionality is offered in the SDK in 3 ways:

1.	Helper functions (in C) inside rsutils (https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L14) allowing the user to project 2D point to 3D space and vice-versa given depth. 
2.	Point-cloud processing-block under processing.hpp that generates 3D points (with texture-coordinates) from depth image
3.	Align processing-block under processing.hpp that generates synthetic depth/color frame from the point-of-view of color/depth imager.

Note that this version of the SDK does not support R200. For R200 support, you have to get the legacy branch. Once in the legacy branch, you should be able to find Projection information here: https://github.com/IntelRealSense/librealsense/blob/legacy/doc/projection.md.

Regards,
Jesus G.
Intel Customer Support

closed 2017-11-27 by jlgarcia75

tchapi, 2017-12-06:
Hi @jlgarcia75 and thanks for the clarification on this point.
Is there a quick example to show how to retrieve RGB values for a specific point ?

I'm looking at the **rs-pointcloud** example that uses OpenGL directly but would like to extract the RGB values for each vertex.
With the legacy librealsense version, I would use the color frame and the color stream intrinsics but it seems that all this has been factored into the pointcloud itself now, though I can't figure out exactly how.

Can you point me in the right direction ?
Thanks !