realsenseai/librealsense #783: Wrapper Python Pointcloud TypeError: No constructor defined!

ananan, 2017-11-07:
| Required Info  |   |TypeError: pyrealsense2.pointcloud: No constructor defined!
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | SR300
| Firmware Version |   | 3.21.0.0
| Operating System & Version |   | ubuntu14.0.4 64bit
| Kernel Version (Linux Only)  |   |4.4.0-98-generic

code : 
import pyrealsense2 as rs
import numpy as np
import cv2

p = rs.pipeline()
p.start()

pc = rs.pointcloud()
frames = p.wait_for_frames()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()
points = pc.calculate(depth_frame)

output:
/usr/bin/python2.7 /home/peter/PycharmProjects/deepcam/test1.py
Traceback (most recent call last):
  File "/home/peter/test1.py", line 10, in <module>
    pc = rs.pointcloud()
TypeError: pyrealsense2.pointcloud: No constructor defined!

Process finished with exit code 1

zivsha, 2017-11-07:
Hi @ananan , thanks for reporting the issue.
I will open a PR to fix this. for now you can save the following to a file and `git apply` it, then rebuild the library. (e.g save to `pybind.patch` and run `git apply pybind.patch`)

```diff
diff --git a/wrappers/python/python.cpp b/wrappers/python/python.cpp
index d6de0d4e..d44594bd 100644
--- a/wrappers/python/python.cpp
+++ b/wrappers/python/python.cpp
@@ -463,8 +463,9 @@ PYBIND11_PLUGIN(NAME) {
                .def("__call__", &rs2::frame_queue::operator());

     py::class_<rs2::pointcloud> pointcloud(m, "pointcloud");
-    pointcloud.def("calculate", &rs2::pointcloud::calculate, "depth"_a)
-              .def("map_to", &rs2::pointcloud::map_to, "mapped"_a);
+    pointcloud.def(py::init<>())
+        .def("calculate", &rs2::pointcloud::calculate, "depth"_a)
+        .def("map_to", &rs2::pointcloud::map_to, "mapped"_a);

     py::class_<rs2::syncer> syncer(m, "syncer");
     syncer.def(py::init<>())
```

ananan, 2017-11-07:
It works ! thank you so much!

closed 2017-11-11 by zivsha