realsenseai/librealsense #784: Librealsense : is it possible to read the D410 temperature when D410 is not streaming

wwppyy, 2017-11-07:
| Required Info  |   |
|---|---|
| Camera Model |  D400 | 
| Firmware Version |  5.8.9 | 
| Operating System & Version |Win10 RS2   |
| Kernel Version (Linux Only)  |   |

We need to check the DS410 both ASIC and Projector temperature  periodically even when the DS410 is not streaming for thermal design related concern.
when checking the code, we found we only can query the temperature when DS410 is streaming..
is this a hard requirement or any way we can read the ASIC and Projector's temperature even D410 is not streaming.
below picture show the temperature query function will throw error if DS410 is not streaming

[image]

wwppyy, 2017-11-14:
@dorodnic , FYI, we have IPDEV installed on our PC and found we can read the temperature by input the command gtemp or mmtemp at its terminal even the module is not streaming.

icarpis, 2017-11-21:
Hello @wwppyy, I checked it and it seems that there isn't any firmware problem/limitation to query the ASIC and Projector's temperature when camera is not streaming. Anyway, we are not going to make this change currently but we will be glad to merge a PR with this change.
Pay attention that if you want to do it by yourself you probably will need to do some changes in 'realsense-viewer' tool and maybe in some other RealSense SDK tools/examples that works with camera options.

closed 2017-11-27 by jlgarcia75