realsenseai/librealsense #785: Detect face using depth/ir camera

ankit-shinde, 2017-11-07:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  SDK2.0 | 
| Operating System & Version |Ubuntu 15.04   |
| Kernel Version (Linux Only)  |   14.014142|

How to distinguish between real face and the one on photo/image.

zivsha, 2017-11-09:
Hi @ankit-shinde ,

This is a really broad question regarding computer vision and face recognition.
The quick and probably unsatisfying answer is that you should use the SDK to get both color images and depth images, and use depth to help you detect real face and images.
I would suggest asking such questions in more relevant forums to get a more professional answer.

closed 2017-12-01 by dorodnic

eebanos, 2018-03-14:
You should focus on Face Presentation Attack Detection, one example is using liveness detection through the calculation of the difference between live and fake faces by using diffusion speed.