realsenseai/librealsense #791: Still getting fourcc INZI error on Ubuntu 16.04

IJKTech, 2017-11-08:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.15.0.0 | 
| Operating System & Version | Ubuntu 16.04 64 bit   |
| Kernel Version (Linux Only)  |  4.13.11-041311-generic |

I am having issues using realsense-viewer, and all of the example scripts since installing librealsense. 

My issue is that I always receive the message:

```
RealSense error calling rs2_pipeline_start(pipe:0x28b05a0):
    Requested pixel format is not natively supported by the Linux kernel and likely requires a patch for fourcc code INZI!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
```

I have updated my kernel to the latest mainline version number.

I've also tried running all of the troubleshooting scripts in the scripts directory. The script `patch-uvcvideo-16.04.simple.sh` appears to have a typo, pointing to a non-existent patch file.

ev-mp, 2017-11-08:
Hi  @IJKTech ,
The installation instructions for Linux backend are quite straight-forward if you follow the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)

The kernel that you've chosen is not Ubuntu LTS for which we provide support, nevertheless I've checked it (4.13.11-generic) with Ubuntu 16.01 and it runs SR300 out-of-the box and does not require additional kernel patches for streaming video. This is the expected behaviour as the SR300-specific formats have been upstreamed into kernel tree since 4.12.

I would recommend you to rerun the installation guide and skip the kernel patch instructions as your kernel is SR300-ready.
Actually, the patch that is mentioned in the guide is going to fail in your case as it covers only the official Ubuntu LTS kernels 4.4/4.8/4.10.

With regards to __patch-uvcvideo-16.04.simple.sh__ this is a deprecated script that is not referenced in the documentation and was a part of the legacy version. We will remove it. Thanks for pointing this out!

IJKTech, 2017-11-09:
Thanks for getting back to me, initially I had followed the installation steps [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) so I was hopeful that I'd be able to get everthing working today.

I purged my system of realsense related binaries and udev rules etc, using the command in the above link, and started fresh.

First I tried with Kernel `4.13.11`. I compiled and installed with examples enabled, skipping the kernel patching, but adding the udev rules. I can see that there is `/dev/video0` and `/dev/video1` (there are no other webcams attached to the machine).

However I still get the same error when running realsense-viewer.

Next I tried with Kernel `4.10.0-38-generic` - the version that was supplied out of the box by my ubuntu install.

This time I also tried patching the kernel. All was going well, modules had been built etc, but then when it came to the section of the script where patches are applied I was greeted with this error:

```
make: Leaving directory '/home/james/librealsense/ubuntu-xenial-hwe'
Patched kernels modules were created successfully

Handle videodev :
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Required key not available
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded 
```

Once again /dev/video0 and /dev/video1 are available when the SR300 is reinserted, but the realsense-viewer error persists.

I tried with a 3rd kernel, 4.4.97.

I get to the same stage, but with a different error message on patch: 

[10:47] 
 ```
Patched kernels modules were created successfully

Handle videodev :
    Module videodev  is in use by uvcvideo
    Unloading dependency uvcvideo
    Module is resident, unloading ...  succeeded. 
Applying the patched module ...  succeeded
    Reloading dependency uvcvideo ... modprobe: ERROR: could not insert 'uvcvideo': Invalid argument
Failed to reload module uvcvideo. error type 1 . Operation is aborted
```

After plugging in the SR300 at this stage, there are no /dev/video devices available.

Please note that in all but the kernel 4.4 case, I can stream the RGB feed via realsense-viewer, or even `cheese`. However my problem arises only when I try to open the *depth* stream.

The full `uname -a` output with kernel 4.10 loaded is: `Linux james-machine 4.10.0-38-generic #42~16.04.1-Ubuntu SMP Tue Oct 10 16:32:20 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

Thanks,
James

ev-mp, 2017-11-09:
Hi, the second and 3rd attempt failed most probably due to discrepancy between the installed Linux image headers file and the kernel running. It would help to see the full log of the patch to verify it.
Please perform the following
- Select and reboot with one of three kernel mentioned
- Connect SR300 while running "dmesg -wT" to monitor enumeration (and post the log)
- Run "modinfo uvcvideo" to check currently loaded version of the driver and post the log
- Get "sudo apt install **v4l-utils**" tool and run "__v4l2-ctl --list-formats-ext -d /dev/video0_", "__v4l2-ctl --list-formats-ext -d /dev/video1__" and post it as well.

IJKTech, 2017-11-09:
With 4.13:

4.13

*Connect SR300 while running "dmesg -wT" to monitor enumeration (and post the log)*

```

[Thu Nov  9 15:17:49 2017] usb 3-2: USB disconnect, device number 2

[Thu Nov  9 15:17:49 2017] usb 3-2.2: USB disconnect, device number 4

[Thu Nov  9 15:17:50 2017] xhci_hcd 0000:00:14.0: Cannot set link state.

[Thu Nov  9 15:17:50 2017] usb usb4-port2: cannot disable (err = -32)

[Thu Nov  9 15:17:50 2017] usb 4-2: USB disconnect, device number 2

[Thu Nov  9 15:17:50 2017] usb 4-2.1: USB disconnect, device number 3

[Thu Nov  9 15:17:52 2017] usb 3-2: new high-speed USB device number 8 using xhci_hcd

[Thu Nov  9 15:17:52 2017] usb 3-2: New USB device found, idVendor=05e3, idProduct=0610

[Thu Nov  9 15:17:52 2017] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0

[Thu Nov  9 15:17:52 2017] usb 3-2: Product: USB2.0 Hub

[Thu Nov  9 15:17:52 2017] usb 3-2: Manufacturer: GenesysLogic

[Thu Nov  9 15:17:52 2017] hub 3-2:1.0: USB hub found

[Thu Nov  9 15:17:52 2017] hub 3-2:1.0: 2 ports detected

[Thu Nov  9 15:17:52 2017] usb 4-2: new SuperSpeed USB device number 4 using xhci_hcd

[Thu Nov  9 15:17:52 2017] usb 4-2: New USB device found, idVendor=05e3, idProduct=0616

[Thu Nov  9 15:17:52 2017] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0

[Thu Nov  9 15:17:52 2017] usb 4-2: Product: USB3.0 Hub

[Thu Nov  9 15:17:52 2017] usb 4-2: Manufacturer: GenesysLogic

[Thu Nov  9 15:17:52 2017] hub 4-2:1.0: USB hub found

[Thu Nov  9 15:17:52 2017] hub 4-2:1.0: 2 ports detected

[Thu Nov  9 15:17:53 2017] usb 3-2.2: new full-speed USB device number 9 using xhci_hcd

[Thu Nov  9 15:17:53 2017] usb 3-2.2: config 1 has an invalid interface number: 3 but max is 2

[Thu Nov  9 15:17:53 2017] usb 3-2.2: config 1 has no interface number 2

[Thu Nov  9 15:17:53 2017] usb 3-2.2: New USB device found, idVendor=041e, idProduct=409b

[Thu Nov  9 15:17:53 2017] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3

[Thu Nov  9 15:17:53 2017] usb 3-2.2: Product: BlasterX Senz3D VF0810

[Thu Nov  9 15:17:53 2017] usb 3-2.2: Manufacturer: Creative Technology Ltd

[Thu Nov  9 15:17:53 2017] usb 3-2.2: SerialNumber: 170222093351

[Thu Nov  9 15:17:53 2017] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.3/0003:041E:409B.0005/input/input19

[Thu Nov  9 15:17:53 2017] hid-generic 0003:041E:409B.0005: input,hiddev0,hidraw1: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:14.0-2.2/input3

[Thu Nov  9 15:17:55 2017] usb 4-2.1: new SuperSpeed USB device number 5 using xhci_hcd

[... 140 lines cut ...]
vermagic:       4.4.90+ SMP mod_unload 

parm:           clock:Video buffers timestamp clock

parm:           hwtimestamps:Use hardware timestamps (uint)

parm:           nodrop:Don't drop incomplete frames (uint)

parm:           quirks:Forced device quirks (uint)

parm:           trace:Trace level bitmask (uint)

parm:           timeout:Streaming control requests timeout (uint)

```

*Get "sudo apt install v4l-utils" tool and run "_v4l2-ctl --list-formats-ext -d /dev/video0", "v4l2-ctl --list-formats-ext -d /dev/video1" and post it as well.*

```

james@james-machine:~$ v4l2-ctl --list-formats-ext -d /dev/video0

Failed to open /dev/video0: No such file or directory

```

IJKTech, 2017-11-09:
With 4.4

dmsg:

```
[Thu Nov  9 15:33:50 2017] usb 4-2: new SuperSpeed USB device number 4 using xhci_hcd
[Thu Nov  9 15:33:50 2017] usb 4-2: New USB device found, idVendor=05e3, idProduct=0616
[Thu Nov  9 15:33:50 2017] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Thu Nov  9 15:33:50 2017] usb 4-2: Product: USB3.0 Hub
[Thu Nov  9 15:33:50 2017] usb 4-2: Manufacturer: GenesysLogic
[Thu Nov  9 15:33:50 2017] hub 4-2:1.0: USB hub found
[Thu Nov  9 15:33:50 2017] hub 4-2:1.0: 2 ports detected
[Thu Nov  9 15:33:51 2017] usb 3-2: new high-speed USB device number 8 using xhci_hcd
[Thu Nov  9 15:33:51 2017] usb 3-2: New USB device found, idVendor=05e3, idProduct=0610
[Thu Nov  9 15:33:51 2017] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Thu Nov  9 15:33:51 2017] usb 3-2: Product: USB2.0 Hub
[Thu Nov  9 15:33:51 2017] usb 3-2: Manufacturer: GenesysLogic
[Thu Nov  9 15:33:51 2017] hub 3-2:1.0: USB hub found
[Thu Nov  9 15:33:51 2017] hub 3-2:1.0: 2 ports detected
[Thu Nov  9 15:33:51 2017] usb 3-2.2: new full-speed USB device number 9 using xhci_hcd
[Thu Nov  9 15:33:52 2017] usb 3-2.2: config 1 has an invalid interface number: 3 but max is 2
[Thu Nov  9 15:33:52 2017] usb 3-2.2: config 1 has no interface number 2
[Thu Nov  9 15:33:52 2017] usb 3-2.2: New USB device found, idVendor=041e, idProduct=409b
[Thu Nov  9 15:33:52 2017] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Nov  9 15:33:52 2017] usb 3-2.2: Product: BlasterX Senz3D VF0810
[Thu Nov  9 15:33:52 2017] usb 3-2.2: Manufacturer: Creative Technology Ltd
[Thu Nov  9 15:33:52 2017] usb 3-2.2: SerialNumber: 170222093351
[Thu Nov  9 15:33:52 2017] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.3/0003:041E:409B.0005/input/input20
[Thu Nov  9 15:33:52 2017] hid-generic 0003:041E:409B.0005: input,hiddev0,hidraw1: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:14.0-2.2/input3
[Thu Nov  9 15:33:52 2017] usbcore: registered new interface driver snd-usb-audio
[Thu Nov  9 15:33:53 2017] usb 4-2.1: new SuperSpeed USB device number 5 using xhci_hcd
[Thu Nov  9 15:33:53 2017] usb 4-2.1: New USB device found, idVendor=8086, idProduct=0aa5
[Thu Nov  9 15:33:53 2017] usb 4-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Thu Nov  9 15:33:53 2017] usb 4-2.1: Product: Intel(R) RealSense(TM) Camera SR300
[Thu Nov  9 15:33:53 2017] usb 4-2.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[Thu Nov  9 15:33:53 2017] media: Linux media interface: v0.10
[Thu Nov  9 15:33:53 2017] videodev: module verification failed: signature and/or required key missing - tainting kernel
[Thu Nov  9 15:33:53 2017] Linux video capture interface: v2.00
[Thu Nov  9 15:33:53 2017] ------------[ cut here ]------------
[Thu Nov  9 15:33:53 2017] WARNING: CPU: 1 PID: 2338 at /home/kernel/COD/linux/kernel/module.c:1127 module_put+0x9e/0xb0()
[Thu Nov  9 15:33:53 2017] Modules linked in: videodev(OE) media snd_usb_audio snd_usbmidi_lib drbg ansi_cprng ctr ccm rfcomm bnep eeepc_wmi asus_wmi sparse_keymap intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp arc4 rtl8821ae kvm snd_hda_codec_hdmi irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel btcoexist snd_hda_codec_realtek rtl_pci snd_hda_codec_generic rtlwifi snd_hda_intel cryptd snd_hda_codec mac80211 snd_hda_core serio_raw snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq cfg80211 snd_seq_device snd_timer snd soundcore input_leds lpc_ich btusb btrtl btbcm btintel bluetooth shpchp mei_me soc_button_array mei mac_hid parport_pc ppdev lp parport autofs4 uas usb_storage hid_generic usbhid hid psmouse i915 ahci i2c_algo_bit libahci drm_kms_helper syscopyarea sysfillrect
[Thu Nov  9 15:33:53 2017]  sysimgblt e1000e fb_sys_fops drm ptp pps_core wmi fjes video
[Thu Nov  9 15:33:53 2017] CPU: 1 PID: 2338 Comm: systemd-udevd Tainted: G           OE   4.4.97-040497-generic #201711081042
[Thu Nov  9 15:33:53 2017] Hardware name: ASUSTeK COMPUTER INC. K30AD_M31AD_M51AD/K30AD_M31AD_M51AD_M32AD, BIOS 0504 02/10/2014
[Thu Nov  9 15:33:53 2017]  0000000000000286 ed1fafe507c4396e ffff8800a1ccbc98 ffffffff813daa23
[Thu Nov  9 15:33:53 2017]  0000000000000000 ffffffff81ccbf78 ffff8800a1ccbcd0 ffffffff810800f2
[Thu Nov  9 15:33:53 2017]  ffffffffc07c7bc0 ffffffffc07c7bc0 ffff8800bd7bcce0 ffffffffc07c7bc0
[Thu Nov  9 15:33:53 2017] Call Trace:
[Thu Nov  9 15:33:53 2017]  [<ffffffff813daa23>] dump_stack+0x63/0x90
[Thu Nov  9 15:33:53 2017]  [<ffffffff810800f2>] warn_slowpath_common+0x82/0xc0
[Thu Nov  9 15:33:53 2017]  [<ffffffff8108023a>] warn_slowpath_null+0x1a/0x20
[Thu Nov  9 15:33:53 2017]  [<ffffffff811048de>] module_put+0x9e/0xb0
[Thu Nov  9 15:33:53 2017]  [<ffffffff8118bc99>] do_init_module+0xd5/0x1cb
[Thu Nov  9 15:33:53 2017]  [<ffffffff8110934a>] load_module+0x161a/0x1ba0
[Thu Nov  9 15:33:53 2017]  [<ffffffff811057f0>] ? __symbol_put+0x60/0x60
[Thu Nov  9 15:33:53 2017]  [<ffffffff81211cd0>] ? kernel_read+0x50/0x80
[Thu Nov  9 15:33:53 2017]  [<ffffffff81109b14>] SYSC_finit_module+0xb4/0xe0
[Thu Nov  9 15:33:53 2017]  [<ffffffff81109b5e>] SyS_finit_module+0xe/0x10
[Thu Nov  9 15:33:53 2017]  [<ffffffff8181b632>] entry_SYSCALL_64_fastpath+0x16/0x71
[Thu Nov  9 15:33:53 2017] ---[ end trace e4839f51808b6d13 ]---
[Thu Nov  9 15:33:53 2017] v4l2_common: disagrees about version of symbol v4l2_subdev_init
[... cut ...]
parm:           clock:Video buffers timestamp clock
parm:           hwtimestamps:Use hardware timestamps (uint)
parm:           nodrop:Don't drop incomplete frames (uint)
parm:           quirks:Forced device quirks (uint)
parm:           trace:Trace level bitmask (uint)
parm:           timeout:Streaming control requests timeout (uint)
```

check devices:

```
james@james-machine:~$ v4l2-ctl --list-formats-ext -d /dev/video0
Failed to open /dev/video0: No such file or directory
```

No devices again this time!

IJKTech, 2017-11-09:
Finally, 4.10, which gives video devices.

dmesg:

```
[Thu Nov  9 15:44:15 2017] usb 4-2: new SuperSpeed USB device number 4 using xhci_hcd
[Thu Nov  9 15:44:15 2017] usb 4-2: New USB device found, idVendor=05e3, idProduct=0616
[Thu Nov  9 15:44:15 2017] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Thu Nov  9 15:44:15 2017] usb 4-2: Product: USB3.0 Hub
[Thu Nov  9 15:44:15 2017] usb 4-2: Manufacturer: GenesysLogic
[Thu Nov  9 15:44:15 2017] hub 4-2:1.0: USB hub found
[Thu Nov  9 15:44:15 2017] hub 4-2:1.0: 2 ports detected
[Thu Nov  9 15:44:15 2017] usb 3-2: new high-speed USB device number 8 using xhci_hcd
[Thu Nov  9 15:44:15 2017] usb 3-2: New USB device found, idVendor=05e3, idProduct=0610
[Thu Nov  9 15:44:15 2017] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Thu Nov  9 15:44:15 2017] usb 3-2: Product: USB2.0 Hub
[Thu Nov  9 15:44:15 2017] usb 3-2: Manufacturer: GenesysLogic
[Thu Nov  9 15:44:15 2017] hub 3-2:1.0: USB hub found
[Thu Nov  9 15:44:15 2017] hub 3-2:1.0: 2 ports detected
[Thu Nov  9 15:44:16 2017] usb 3-2.2: new full-speed USB device number 9 using xhci_hcd
[Thu Nov  9 15:44:16 2017] usb 3-2.2: config 1 has an invalid interface number: 3 but max is 2
[Thu Nov  9 15:44:16 2017] usb 3-2.2: config 1 has no interface number 2
[Thu Nov  9 15:44:16 2017] usb 3-2.2: New USB device found, idVendor=041e, idProduct=409b
[Thu Nov  9 15:44:16 2017] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Nov  9 15:44:16 2017] usb 3-2.2: Product: BlasterX Senz3D VF0810
[Thu Nov  9 15:44:16 2017] usb 3-2.2: Manufacturer: Creative Technology Ltd
[Thu Nov  9 15:44:16 2017] usb 3-2.2: SerialNumber: 170222093351
[Thu Nov  9 15:44:16 2017] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.3/0003:041E:409B.0005/input/input19
[Thu Nov  9 15:44:16 2017] hid-generic 0003:041E:409B.0005: input,hiddev0,hidraw1: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:14.0-2.2/input3
[Thu Nov  9 15:44:18 2017] usb 4-2.1: new SuperSpeed USB device number 5 using xhci_hcd
[Thu Nov  9 15:44:18 2017] usb 4-2.1: New USB device found, idVendor=8086, idProduct=0aa5
[Thu Nov  9 15:44:18 2017] usb 4-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Thu Nov  9 15:44:18 2017] usb 4-2.1: Product: Intel(R) RealSense(TM) Camera SR300
[Thu Nov  9 15:44:18 2017] usb 4-2.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[Thu Nov  9 15:44:18 2017] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[Thu Nov  9 15:44:18 2017] uvcvideo 4-2.1:1.0: Entity type for entity Processing 2 was not initialized!
[Thu Nov  9 15:44:18 2017] uvcvideo 4-2.1:1.0: Entity type for entity Camera 1 was not initialized!
[Thu Nov  9 15:44:18 2017] uvcvideo 4-2.1:1.0: Entity type for entity Extension 3 was not initialized!
[Thu Nov  9 15:44:18 2017] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.1/4-2.1:1.0/input/input20
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 5a564e49-2d90-4a58-920b-773f1f2c556b
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 495a4e49-1a66-a242-9065-d01814a8ef8a
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 49564e49-57db-5e49-8e3f-f479532b946f
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[Thu Nov  9 15:44:18 2017] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[Thu Nov  9 15:44:18 2017] uvcvideo: Unable to create debugfs 4-5 directory.
[Thu Nov  9 15:44:18 2017] uvcvideo 4-2.1:1.2: Entity type for entity INV_EU_STRING was not initialized!
[Thu Nov  9 15:44:18 2017] uvcvideo 4-2.1:1.2: Entity type for entity Camera 5 was not initialized!
[Thu Nov  9 15:44:18 2017] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.1/4-2.1:1.2/input/input21
```

modinfo:

```
james@james-machine:~$ modinfo uvcvideo
filename:       /lib/modules/4.10.0-38-generic/kernel/drivers/media/usb/uvc/uvcvideo.ko
version:        1.1.1
license:        GPL
description:    USB Video Class driver
author:         Laurent Pinchart <laurent.pinchart@ideasonboard.com>
srcversion:     A5BB8A9B6C1A8F92A5DA753
alias:          usb:v*p*d*dc*dsc*dp*ic0Eisc01ip01in*
alias:          usb:v*p*d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v2833p0211d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v2833p0201d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1C4Fp3000d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1B3Bp2951d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d00*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d01[0-1]*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d012[0-6]dc*dsc*dp*ic0Eisc01ip00in*
[... 160 lines cut ...]
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

	Index       : 5
	Type        : Video Capture
	Pixel Format: ''
	Name        : 52494150-8536-4841-b6bf-8fc6ffb
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x240
			Interval: Discrete 0.009s (110.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)

```

IJKTech, 2017-11-09:
I posted [this question](https://unix.stackexchange.com/questions/403270/patching-fourcc-code-inzi?noredirect=1#comment721405_403270) yesterday, and get a reply earlier from a person who had updated to 4.12 and resolved the issue. 

The difference is that their camera has firmware 3.21, whilst mine has 3.15, could that be the underlying problem?

ev-mp, 2017-11-12:
Hi @IJKTech ,
In the `dmesg ` log for 4.13 it says:
`[Thu Nov  9 15:17:55 2017] uvcvideo: version magic '4.4.90+ SMP mod_unload ' should be '4.13.11-041311-generic SMP mod_unload ' which means that your compiled patch with headers of kernel 4.4 while running kernel 4.13.
Resetting/reinstalling the original 4.13 kernel with its original drivers will fix it

With kernel 4.4 log -  are you using Ubuntu LTS kernel ? Which should have **4.4.0-xx** pattern, but in the log it reports **4.4.97** which is more appropriate for the vanilla kernels, subsequently when trying to load the kernel issues multiple warnings:
`v4l2_common: disagrees about version of symbol v4l2_subdev_init ...`

For kernel 4.10 log it seem that it is set correctly, but it hasn't been patches with Librealsense instructions, so SR300 specific formats are not recognized:
`[Thu Nov  9 15:44:18 2017] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.1/4-2.1:1.0/input/input20
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 5a564e49-2d90-4a58-920b-773f1f2c556b
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 495a4e49-1a66-a242-9065-d01814a8ef8a
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 49564e49-57db-5e49-8e3f-f479532b946f
[Thu Nov  9 15:44:18 2017] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8`

It is not clear from your statements whether you have this resolved, but if not - 
please stick to 4.10 which is the latest Ubuntu LTS version, run through the installation guide including patching the kernel and applying udev permissions, and then update here.

dorodnic, 2017-11-22:
Closing the issue for now. @IJKTech feel free to provide more information and re-open if you encounter any problems with the steps suggested by @ev-mp

closed 2017-11-22 by dorodnic

akashjinandra, 2018-01-16:
Hello, I seem to be having the same problem, I installed using dpkg, and then from source.  After running uname -r I have: 4.13.0-26-generic. Also running on an intel NUC

15/01 18:10:51,294 ERROR [140255151770304] (types.h:152) T4.13.0-26-generic
he requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device

peteflorence, 2018-01-26:
Hello, also having same issue.

Here are some very relevant notes on system configuration:
- 4.13.0-31-generic
- Have gone through all steps (afaik) for linux patch
- Using librealsense-2.9.0
- Using realsense_ros_camera 2.0.1
- Launches the D435 just fine
- The SR300 fails with this error

Also note:
- With apt-getted realsense_camera, able to launch the SR300 but the depth image flickers terribly, unsable

Also note:
- Confirmed that the two SR300s we are using work just fine on a different computer, also on 4.13.0-31-generic, unknown what is different

Would be very helpful to have more of a sense what is causing different SR300 failure modes.  Thanks!

CKunz, 2018-02-02:
Hello,

I have the same problem as described above.

- 4.13.0-32-generic
- SR 300
- librealsense-2.9.0

error-log from realsense-viewer:
```
realsense-viewer 
 02/02 13:47:04,645 WARNING [140206066546432] (backend-v4l2.cpp:1100) Pixel format 5a564e49-2d90-4a58-920b-773f1f2 likely requires patch for fourcc code INVZ!
 02/02 13:47:04,645 WARNING [140206066546432] (backend-v4l2.cpp:1100) Pixel format 495a4e49-1a66-a242-9065-d01814a likely requires patch for fourcc code INZI!
 02/02 13:47:04,646 WARNING [140206066546432] (backend-v4l2.cpp:1100) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 02/02 13:47:04,646 WARNING [140206066546432] (backend-v4l2.cpp:1100) Pixel format 49564e49-57db-5e49-8e3f-f479532 likely requires patch for fourcc code INVI!
 02/02 13:47:04,646 WARNING [140206066546432] (backend-v4l2.cpp:1100) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 02/02 13:47:04,647 WARNING [140206066546432] (sensor.cpp:313) Unregistered Media formats : [ INVZ INZC PAIR YUYV ]; Supported: [ ]
 02/02 13:47:16,469 WARNING [140206347582208] (archive.cpp:315) The user was holding on to 2 frames after stream 0x0x3008ea0 stopped
 02/02 13:47:16,470 WARNING [140206347582208] (archive.cpp:325) All frames from stream 0x0x3008ea0 are now released by the user
 02/02 13:47:17,087 ERROR [140206347582208] (types.h:178) The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
 02/02 13:53:21,823 ERROR [140206347582208] (types.h:178) The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
```

I went through the standard installation procedure and also build it on my machine.

Any guess?

JesusCoyotzi, 2018-02-06:
Have the same problem.

- Linux 4.13.-32-generic
- SR300
- librealsense-2.9

My log for reference:
```
 06/02 13:24:31,049 WARNING [139643918972672] (backend-v4l2.cpp:1100) Pixel format 5a564e49-2d90-4a58-920b-773f1f2 likely requires patch for fourcc code INVZ!
 06/02 13:24:31,050 WARNING [139643918972672] (backend-v4l2.cpp:1100) Pixel format 495a4e49-1a66-a242-9065-d01814a likely requires patch for fourcc code INZI!
 06/02 13:24:31,050 WARNING [139643918972672] (backend-v4l2.cpp:1100) Pixel format 435a4e49-b602-480f-978c-e4e88ae likely requires patch for fourcc code INZC!
 06/02 13:24:31,050 WARNING [139643918972672] (backend-v4l2.cpp:1100) Pixel format 49564e49-57db-5e49-8e3f-f479532 likely requires patch for fourcc code INVI!
 06/02 13:24:31,051 WARNING [139643918972672] (backend-v4l2.cpp:1100) Pixel format 52494150-8536-4841-b6bf-8fc6ffb likely requires patch for fourcc code PAIR!
 06/02 13:24:31,051 WARNING [139643918972672] (sensor.cpp:313) Unregistered Media formats : [ INVZ INZC PAIR YUYV ]; Supported: [ ]
 06/02 13:24:34,980 ERROR [139644182334208] (types.h:178) The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
```