realsenseai/librealsense #792: Different performance of rs-align example / code

wwangzhao, 2017-11-09:
| Required Info  |   |
|---|---|
| Camera Model |  SR300  | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |   |

Hi 
Thanks for the help of previous issue
Now I have start to synchronize the depth and rgb channel

I have complied the code. Compare with the rs-align examples that build while installing, my compiled project works but the window always blink, which seems that the frame rate/updating frequency is not good? The rs-align example works much more stable

Regards
Zhao

zivsha, 2017-11-09:
Hi,
Are you using `example.hpp` and its `window` class for the rendering loop?
If so then please note that the `while (app) { /*...*/ }`  both swaps and clears the screen buffer each iteration, this is because `window` performs all of the OpenGL code inside its boolean casting operator:

https://github.com/IntelRealSense/librealsense/blob/a8b8567f3859f7c1cf1ff557fa03b1d22b322376/examples/example.hpp#L177-L197

This means that if you have some `continue` inside the loop, which continues to the next iteration before "uploading" the next image, the loop will render a blank frame and not the previous image.

If this is not the case, please share some of your code so we can try to assist.

wwangzhao, 2017-11-09:
Hi @zivsha 

It sounds like the reason u posted there. I use the example.hpp

> #include <librealsense2/rs.hpp>     // Include RealSense Cross Platform API
> #include"example.hpp"
> #include<imgui.h>
> #include"imgui_impl_glfw.h"
> 
> #include<sstream>
> #include<iostream>
> #include<fstream>
> #include<algorithm>
> #include<cstring>
> 
> // where all the GUI code goes
> void render_slider(rect location, float& clipping_dist);
> 
> // takes depth and color images, the depth scale units, and the maximum distance the user wishes to show,
> // and updates the color frame so that its background (any pixel with depth distance larger than the maximum allowed)
> // is removed.
> void remove_background(rs2::video_frame& color, const rs2::depth_frame& depth_frame,
>                        float depth_scale, float clipping_dist);
> 
> // find a depth sensor from the pipeline's device and retrieves its depth scale units.
> float get_depth_scale(rs2::device dev);
> 
> // goes over the given streams and verify that it has a depth profile
> // and tries to find another profile to which depth should be aligned.
> rs2_stream find_stream_to_align(const std::vector<rs2::stream_profile>& streams);
> 
> // checks if the current streaming profiles contains all the previous one
> bool profile_changed(const std::vector<rs2::stream_profile>& current, const std::vector<rs2::stream_profile>& prev);
> int main(int argc, char * argv[]) try
> {
>     window app(1280,720,"WZ's Realsense Capture Align");
>     ImGui_ImplGlfw_Init(app, false);
>     texture renderer;
>     rs2::colorizer color_map;
>     window app(1280,720,"WZ's Realsense Capture Align");
>     ImGui_ImplGlfw_Init(app, false);
>     texture renderer;
>     rs2::colorizer color_map;
>     rs2::pipeline pipe;
> 
>     // Configure and start the pipeline
>     rs2::pipeline_profile profile = pipe.start();
> 
>     float depth_scale = get_depth_scale(profile.get_device());
> 
>     rs2_stream align_to = find_stream_to_align(profile.get_streams());

>     rs2::align align(align_to); 
> 
>     // Define a variable for controlling the distance to clip
>     float depth_clipping_distance = 1.f;
>     // Create a Pipeline - this serves as a top-level API for streaming and processing frames
>     rs2::pipeline pipe;
>     // Configure and start the pipeline
>     rs2::pipeline_profile profile = pipe.start();
> 
>     float depth_scale = get_depth_scale(profile.get_device());
> 
>     rs2_stream align_to = find_stream_to_align(profile.get_streams());
> 
>     // Create a rs2::align object
> 
>     rs2::align align(align_to); 
> 
>     // Define a variable for controlling the distance to clip
[... 119 lines cut ...]
>     // Using the pipeline's profile, we can retrieve the device that the pipeline uses
>     float depth_scale = get_depth_scale(profile.get_device());
> 
>     //Pipeline could choose a device that does not have a color stream
>     //If there is no color stream, choose to align depth to another stream
>     rs2_stream align_to = find_stream_to_align(profile.get_streams());
> 
>     // Create a rs2::align object
>     // rs2::align allows us to perform alignment of depth frames to others frames
>      // rs2::align allows us to perform alignment of depth frames to others frames
>     rs2::align align(align_to); // The "align_to" is the stream type to which we plan to align depth frames.
> 
>     // Define a variable for controlling the distance to clip
>     float depth_clipping_distance = 1.f;25);
> 
>         // Render depth (as picture in pipcture)
>         renderer.upload(color_map(aligned_depth_frame));
>         renderer.show(pip_stream);
> 
>         // Using ImGui library to provide a slide controller to select the depth clipping distance
>         ImGui_ImplGlfw_NewFrame(1);
>         render_slider({ 5.f, 0, w, h }, depth_clipping_distance);
>         ImGui::Render();
> }
>

zivsha, 2017-11-09:
@wwangzhao can you please narrow the code snippet to only the differences between your code and the `rs-align` example?

Generally you can do one of the following:
1. Not use `while (app)` and instead use `while (some_condifion)` only flush the buffer when you need.
2. Make sure you continue to next iteration only after rendering the next image
3. Always save the last rendered images and if you don't get new frames, render the previous ones.
4. Any other way that makes sure the flow does not render an empty buffer.

Please note that the example code is a just a reference and quick starting point. 
For more complex programs (which I'm not sure yet that your code is such), I would suggest that you do not rely on `example.hpp`.

wwangzhao, 2017-11-09:
Hi @zivsha 

Thanks a lot for your help. There is no difference between my code and the rs-align example code provided in 

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

If possible would u please provide an example of 1. ?

Regards
zhao

closed 2018-02-03 by zivsha