realsenseai/librealsense #793: Unable to import pyrealsense2 on python

alexanderPeregrina, 2017-11-09:
| Required Info  |   |
|---|---|
| Camera Model | SR300  | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.10.0-38-generic |

I can't import pyrealsense library on python, I tried pip install pyrealsense, but a an error occur:

pyrealsense/rsutilwrapper.cpp:572:16: fatal error: rs.h: No existe el archivo o el directorio
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

Any suggestion with this?... 
I hope you can help me

zivsha, 2017-11-09:
Hi @alexanderPeregrina ,
Which version are you using? `rsutilwrapper.cpp` looks like legacy code

alexanderPeregrina, 2017-11-09:
When I use pip install pyrealsense is trying to install version 2.2 of pyrealsense, I've already install librealsense version 2 and the firmware version of my intel realsense sr300 is 3.21.0.0, Any suggestion?, I need to work with this camera and open CV python, I hope you can help me.

alexanderPeregrina, 2017-11-10:
I run pip install pyrealsense as root, and the last error just disappear, but I have an importing error:

File "/tmp/pip-build-fU769y/pyrealsense/setup.py", line 9, in <module>
        from Cython.Build import cythonize
    ImportError: No module named Cython.Build
Any suggestion?

zivsha, 2017-11-11:
Hi @alexanderPeregrina,
I could figure out from your previous answer which version of the SDK you are using. Versions are available under the [releases tab](https://github.com/IntelRealSense/librealsense/releases).
Can you please elaborate on what was the installation process you did? Are you trying to only run `pip install` without anything else?

alexanderPeregrina, 2017-11-12:
HI, I've got SDK version 2.8.1, I've installed librealsense and run few examples, then I went to the wrapper folder and I followed this ttps://github.com/IntelRealSense/librealsense/tree/development/wrappers/python, I ran cmake with this flag -DBUILD_PYTHON_BINDINGS=bool:true, then I went to this page https://pypi.python.org/pypi/pyrealsense/2.0 and I installed pycparser, cython and numpy, I downloaded the tar file, I extracted on Downloads folder and I tried python setup.py install, and the following error appears: pyrealsense/rsutilwrapper.cpp:477:16: fatal error: rs.h: doesn't exit file or directory
 #include "rs.h"

What I did wrong?, I hope you can help me.

zivsha, 2017-11-12:
Hi @alexanderPeregrina,
First of, cool photo using our viewer 🤘 .
Second, I must say that pypi/pyrealsense is new to me and I was not aware of it until now.
This is a community supported addition which from the looks of it, doesn't support the new version (see [PR#55](https://github.com/toinsson/pyrealsense/issues/55)). I offered the author my assistance.

alexanderPeregrina, 2017-11-13:
So, is not possible to use python wrapper yet for the version 2 of librealsense?

zivsha, 2017-11-13:
Sorry, I didn't mean to confuse you.
The python wrapper we does work it is just not possible to install it via `pip`.
To use it you simply need to get the compiled output of the wrapper and use it as a python library.
If you say you built it already, just try to run one of the examples under wrappers/python/examples.
Or simply run python and try to `impor pyrealsense` and make sure you don't get an error from that line, if you don't, you can now work with the wrapper,if you do- please post the issue here...

alexanderPeregrina, 2017-11-13:
ok, but what is the directory where I must apply cmake  -DBUILD_PYTHON_BINDINGS=bool:true?
Is librealsense/wrappers/python?

zivsha, 2017-11-15:
No, the you should pass this argument when running the main CMake of the project.
For example:
```bash
git clone https://github.com/IntelRealSense/librealsense
cd librealsense
mkdir build
cd build
cmake ../ -DBUILD_PYTHON_BINDINGS=TRUE
make -j4
sudo make install #Optional if you want the library to be installed in your system
```
If it all goes without errors, you should be able to find the `pyrealsense2.<arch info>.so` under `build/wrappers/python` (actually 3 files with the same name and extensions `.so`, `.so.2`, `.so.2.8.1`). 
Now the easiest way to use it is run python from that folder and `import pyrealsense2`

alexanderPeregrina, 2017-11-16:
It works fine, thanks, but, it works for python3, how can I rebuild for python 2.7?..

zivsha, 2017-11-16:
If you have multiple versions of python on your machine:
From [python/readme.md](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md):
> **Note**: To force compilation with a specific version on a system with both Python 2 and Python 3 installed, add the following flag to CMake command:
`-DPYTHON_EXECUTABLE=[full path to the exact python executable]`

alexanderPeregrina, 2017-11-16:
It works good, Thank you very much!

closed 2017-11-16 by alexanderPeregrina

kazoo-kmt, 2018-01-08:
Hi @zivsha, I'm trying to install `pyrealsense2` on Mac OS. Based on [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md#building-from-source), I installed the necessary packages and I ran 
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS=true -G Unix\ Makefiles`
instead of 
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -G Xcode`.

Now under `/usr/local/lib`, I can see the followings.
`pyrealsense2.2.9.0.cpython-36m-darwin.so`
`pyrealsense2.2.cpython-36m-darwin.so`
`pyrealsense2.cpython-36m-darwin.so`

Also, I set `export LD_LIBRARY_PATH=/usr/local/lib` and `export PYTHONPATH=$PYTHONPATH:/usr/local/lib`. 

When I call `import pyrealsense2` from the python interpreter, it says `ImportError: No module named 'pyrealsense2'`. 

What should I try to import `pyrealsense2`?

kazoo-kmt, 2018-01-09:
Never mind. After I specified like`-DPYTHON_EXECUTABLE=[full path to the exact python executable]`, I could import.

GitwellAnyohub, 2018-04-19:
can anyone @kazoo-kmt @zivsha tell me what full path to the exact python executable means?

GitwellAnyohub, 2018-04-19:
nvm i figured it out. Sorry!
just do:
import sys
print sys.executable

kat-mcg-lin, 2019-05-27:
can someone tell me exactly what they coded to get this to work? I've tried following your steps and the pyrealsense2 library won't import to use in pycharm

like, after doing the cmake command should I just be able to import pyrealsense2? is there something i missed?

lramati, 2019-06-06:
You need to make sure that the binaries are discoverable by python. The easiest ways to do this are to either copy them to the folder you are running your python script from, or to copy them to the `site-packages` directory for python

jamjahal, 2020-08-16:
Which are the binary files that need to be transfered to `site-packages`?