realsenseai/librealsense #798: [Q] ZR300 IMU model?

cynosure4sure, 2017-11-10:
Does anyone know the manufacturer and model of IMU used inside ZR300? Trying to find IMU datasheet. Thank you.

xjtao1314, 2017-11-22:
I also wanna to know how to get the variance of acc and gyro? I get different result with zr300, but the variance of zr300 seems more accurate; by the way , what unit does zr300 use for variance?

cynosure4sure, 2017-11-23:
I couldn't find the model or parameters of the IMU. So I estimated IMU parameters using Alan Variance plots, my results are as follows:
Accelerometer:

X-axis
Noise Density:  0.001406
Random Walk:    0.00008
Y-axis
Noise Density:  0.0008624
Random Walk:    0.0004539
Z-axis
Noise Density:  0.001115
Random Walk:    0.00004

Gyroscope:
X-axis
Noise Density:  9.508e-05
Random Walk:    3e-06
Y-axis
Noise Density:  0.0001544
Random Walk:    1e-05
Z-axis
Noise Density:  8.529e-05
Random Walk:    4e-06

Looking at my data I think something is not right with y-axis estimates as they x and z axis estimates seem to be in agreement. Looking at the parameters I think the IMU used is a cheap one and I don't know if any low pass filtering done on-board the IMU so I would increase each estimate by a factor of at least 10 before using.
I followed the process and units listed here:
https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model

xjtao1314, 2017-11-23:
I also draw allan variance plot today,  and got the result below:
X-axis
Noise Density: 0.0019
Random Walk: 0.000183
Y-axis
Noise Density: 0.00106
Random Walk: 0.00029
Z-axis
Noise Density: 0.00138
Random Walk: 0.000233

but from zr300, the variance for acc is  0.147 for all axises. and 0.147 seems more accurate to my vio system. 
 
does any one have idea for correct calib IMU variance?
and the plot as follows:

[image]

[image]

xjtao1314, 2017-11-23:
I wander if the max variance in the plot should be choosed . do you get good result with this variance?
[image]
[image]
[image]

cynosure4sure, 2017-11-25:
I would say I get "okay" results but not as good as any other IMU. I might switch to using an external IMU as this internal IMU is not synced with camera frames anyways. I am also attaching my alan standard deviation plots for your comparison. Which software did you use to draw your plots? I used a MATLAB script found online.

[image]

[image]

xjtao1314, 2017-11-27:
I am using https://github.com/GAVLab/allan_variance( changed for better outlook). And a recent inspiration is : maybe we should choose allan variance at tau = 0.008 and slope = -1/2, refer to " and  correspond to the values at  (point (1) in the figure below). This is only true since the noise power in most inertial sensors is dominated by "white noise" at a frequency of approximately 1Hz." (from https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model) . In my opinion, the white noise has almost the same frequency with the acc frequency(250 HZ for ZR300). If so , I got almost the same variance with ZR300 provided.

And BTW, you mentioned the sync between IMU and fisheye is not good in ZR300？ how do you get the result?  I assumed the sync is good in my study  .

cynosure4sure, 2017-11-27:
Oh I just remembered one thing. For ZR300 accelerometer runs at 250Hz and Gyro runs at 200Hz but if you see the data from ROS driver it is published at 400Hz i.e. as soon as accelerometer or gyro measurement arrives it copies the other one's data and publishes it. So messages can have repeat data. I modified the driver to publish at 200Hz @ rate of gyro and copy the closest accelerometer measurement, this way although I lose one accelerometer measurement for every 5 measurements still there is no repeated data, maybe you have modified it similarly as well.
When I mentioned sync, I meant no sync between IMU and Color/depth camera, as I am not using fisheye camera for now.

jlgarcia75, 2017-11-27:
Hello @cynosure4sure,

I finally found the IMU model in the ZR300. It is, Bosch BMI055.

Regards,
Jesus G.
Intel Custome Support

closed 2017-11-28 by jlgarcia75