realsenseai/librealsense #802: pipe.poll_for_frames(&frames) only gives us a 10 frames

pwspatz, 2017-11-10:
| Camera Model | SR300 | 
| Firmware Version | newest  | 
| Operating System & Version |  Linux 16.04 Xubuntu |
| Kernel Version (Linux Only)  | 4.10.38  |

The function pipe.poll_for_frames(&frames) gives us only 10 frames. 
Perhaps it has something to do with: 
"However, if the application maintains too long history, the device may lack memory resources to produce new frames, and the following calls to this method shall return no new frames, until resources are retained."?

zivsha, 2017-11-12:
@pwspatz , is this what your application is doing? are you holding the frames for too long?

ntm97, 2017-11-12:
i basically have the same problem too. what exactly does "holding the frames for too long mean"? as soon as i get a frameset i retrieve a point cloud from the depth frame and then send that out to a completely separate part of the application where it will be used for further processing. all of this works completely fine wihtout causing any lack of memory resources when i use wait_for_frames() to retrieve the frames but somehow this method causes  more than 600% (on 8 threads) cpu utilization on an i7 quadcore processor. also i was never able to reach the 60 fps mark (more like 37 - 50 fps). thats why i wanted to try poll_for_frames() instead but somehow it just gives me about 10-13 frames and then it starts to just always return false for as long as the application keeps running.

dorodnic, 2017-12-14:
This seems rather odd. I think a code snippet is necessary to understand the problem.
Even if processing is taking long time, `poll_for_frames` will continue to provide new frames once the initial 10-13 get processed. So it seems like frames are not being released (stored in some data structure perhaps or improperly used from another language).
You can also build the SDK with CMake flag `-DTRACE_API=true` to follow the lifetime of each individual frame inside librealsense log.

Alonmeytal, 2017-12-26:
@dorodnic I think this discussion is rather similar to my question (#946 ) about not being to poll more than 15 frames before "stopping". If you could maybe extend your documentation about your memory management, it might solve everyone's problems by setting their expectations right, to begin with.

@pwspatz I am using the python writing, but the way I solved it by first converting the "color_frame.data" to a multidimensional array (using numpy from code example) and then cloning that array (as the python data structure for color_frame.data is not clonable).
This was apparently enough for me to convince the SDK am "no longer using those frame", and it could recycle them for new frames.
I hope it helps.

dorodnic, 2018-01-18:
Please check out [`frame::keep()`](https://github.com/IntelRealSense/librealsense/blob/development/include/librealsense2/h/rs_frame.h#L188) method that was added to help in this situation.
We have also added a [new example](https://github.com/IntelRealSense/librealsense/tree/development/examples/measure) demonstrating how one might do long processing on a background thread. 
I still plan to expand the documentation (perhaps add to Wiki) on this subject as was suggested.

closed 2018-01-23 by dorodnic