realsenseai/librealsense #803: Doxyfile INPUT too broad

callendorph, 2017-11-10:
| Required Info  |   |
|---|---|
| Camera Model | N/A | 
| Firmware Version | N/A  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0 |

In the doxy configuration file "doc/doxygen/doxyfile" you have the following configuration: 

```
INPUT                  = ../../include \
                         ../../include/librealsense \
                         ../ \
                         ../../src
```

In the file "doc/readme.md", to build the documentation it says to run: 
```
sudo apt-get install doxygen
doxygen doc/doxygen/doxyfile
```

It is not explicitly stated, but it seems the assumption would be to run this command from the root project directory. If you do this, the INPUT definitions point to the parent directory of the root project directory, bringing in anything found there as well. So if you have other opensource projects in that directory, the doxygen documentation also brings in things that are completely unrelated.

So instead, I ran from the "doc" subdirectory of the project. This seems to correctly find the sources for building the project. The only INPUT definition that actually matches is the '../' definition. Unfortunately, if you have patched the kernel prior to running this command, then the kernel sources will be in subdirectory of the project ("ubuntu-xenial-master" in my case). Doxygen then attempts to build all of the documentation for the entirety of the kernel sources. This quickly caused the small instance I was using to run out of memory. It seems unlikely to me that you would intentionally want to build doxygen for the entirety of the linux kernel.

I think the INPUT definition needs to be changed to avoid building the kernel's docs.

ChrisH-Intel, 2017-11-10:
Zendesk ticket #[440](https://intel.zendesk.com/agent/tickets/440) has been linked to this issue.

dorodnic, 2017-11-22:
Hi @callendorph 
Sorry for delayed response. I believe you are correct. We are planning to run doxygen automatically on our servers and host proper output using github pages in the upcoming weeks. In the meanwhile, you are welcomed to contribute changes to the file and documentation that worked for you.

RealSense-Customer-Engineering, 2018-05-24:
[Realsense Customer Engineering Team Comment]
@callendorph  Considering this issue fixed already, can we close it now?

closed 2018-06-05 by RealSense-Customer-Engineering