realsenseai/librealsense #807: Find the Camera , but cant connect to it?

EdXian, 2017-11-11:
Hi all,
Today,I try to connect my R200 camera to Ubuntu PC and I got the following error after running rs-* command . I had used it successfully with Intel ROS package `ROS-Camera` and I could see image by adding image rostopics. It seems that the firmware of my R200 is too old to use.
How do I update the firmware of my camera with an Ubuntu PC?
Any Idea ?
```
 There are 1 connected RealSense devices.

Using device 0, an Platform Camera
    Serial number: 1-1.1-3
    Firmware version: (null)

rs_error was raised when calling rs2_get_device_info(dev:0x1f9ec50, info:Firmware Version):
    info Firmware Version not supported by the device!
```
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | Null  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.10.0-38-generic |

ev-mp, 2017-11-12:
Hello, @EdXian , 
You're running Librealsense2 SDK with R200 which are not compatible - see the [overview](https://github.com/IntelRealSense/librealsense/blob/master/readme.md#overview).
https://github.com/IntelRealSense/librealsense#32
R200 is continues to be maintained via the [legacy](https://github.com/IntelRealSense/librealsense/tree/legacy) branch though.

The reason that the librealsense2 reported ` There are 1 connected RealSense devices.` is because it found and registered an additional webcam/uvc device that you have attached to your system, which the library supports via `Platform Camera` tag, but not the R200 as you might thought.

EdXian, 2017-11-12:
Hi @ev-mp ,
Thank you for your help!!
I have changed my branch to `legacy` and run the following command : 
```
mkdir build && cd build
cmake .. -DBUILD_EXAMPLE=true
make
sudo make install
```
I couldnt find any executable exmaple in the source or usr/bin .Then I went into the folder `examples`  and ran the following command:
```
mkdir build && cd build
make
sudo make install
```
I got the error as following:
```
edxian@edxian:~/librealsense/examples/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
CMake Error at CMakeLists.txt:123 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target
  "c-tutorial-3-pointcloud".

CMake Error at CMakeLists.txt:145 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target
  "cpp-tutorial-1-depth".

-- Configuring incomplete, errors occurred!
See also "/home/edxian/librealsense/examples/build/CMakeFiles/CMakeOutput.log".
```
Thanks.

ev-mp, 2017-11-12:
Hi @EdXian , you should invoke `cmake ../ && make && sudo make install` from the librealsense root folder only.
Please follow the console at the end of  `sudo make install` to see where the binaries being distributed after successful compilation. The convention is actually to put under `/usr/local/bin` rather than `/usr/bin`
Post the console log here if you need more assistance.

EdXian, 2017-11-12:
Hi @ev-mp ,
Thank you very much!
I got the following message in my terminal:
```
edxian@edxian:~/librealsense/build$ cmake ../ && make && sudo make install
-- Info: REALSENSE_VERSION_STRING=1.12.3
-- Building in a ROS environment
-- Using CATKIN_DEVEL_PREFIX: /home/edxian/librealsense/build/devel
-- Using CMAKE_PREFIX_PATH: /home/edxian/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/edxian/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/edxian/librealsense/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- Configuring done
-- Generating done
-- Build files have been written to: /home/edxian/librealsense/build
[ 61%] Built target realsense
[ 67%] Built target SR300-live-test
[ 75%] Built target R200-live-test
[ 80%] Built target F200-live-test
[ 88%] Built target ZR300-live-test
[ 92%] Built target offline-test
[100%] Built target LR200-live-test
[sudo] password for edxian: 
[ 61%] Built target realsense
[ 67%] Built target SR300-live-test
[ 75%] Built target R200-live-test
[ 80%] Built target F200-live-test
[ 88%] Built target ZR300-live-test
[ 92%] Built target offline-test
[100%] Built target LR200-live-test
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/_setup_util.py
-- Up-to-date: /usr/local/env.sh
-- Up-to-date: /usr/local/setup.bash
-- Up-to-date: /usr/local/setup.sh
-- Up-to-date: /usr/local/setup.zsh
-- Up-to-date: /usr/local/.rosinstall
-- Installing: /usr/local/lib/pkgconfig/librealsense.pc
-- Up-to-date: /usr/local/share/librealsense/cmake/librealsenseConfig.cmake
-- Up-to-date: /usr/local/share/librealsense/cmake/librealsenseConfig-version.cmake
-- Up-to-date: /usr/local/share/librealsense/package.xml
-- Up-to-date: /usr/local/lib/librealsense.so.1.12.3
-- Up-to-date: /usr/local/lib/librealsense.so.1
-- Up-to-date: /usr/local/lib/librealsense.so
-- Up-to-date: /usr/local/include/librealsense
-- Up-to-date: /usr/local/include/librealsense/rscore.hpp
-- Up-to-date: /usr/local/include/librealsense/rs.hpp
-- Up-to-date: /usr/local/include/librealsense/rs.h
-- Up-to-date: /usr/local/include/librealsense/rsutil.h
```

ev-mp, 2017-11-12:
You skipped `-DBUILD_EXAMPLES=true` flag in your post.
Without this flag it only builds the shared object and unit-tests.
Rerun the build with the flag , and expect to have the binaries built into the `/usr/local/bin` dir.

EdXian, 2017-11-12:
Hi @ev-mp ,
Thank you very much. It works fine.

EdXian, 2017-11-12:
Hi @ev-mp ,
Sorry for that I got an another problem after closing the issue . 
I run the command  ` cpp-*` and get an error message:
```
cpp-capture: error while loading shared libraries: librealsense.so.1: cannot open shared object file: No such file or directory
```

ev-mp, 2017-11-12:
Hi @EdXian , check whether the file is registered with Ubuntu by `which librealsense.so` - if not present then run `ldconfig` to refresh the so files cache, or relogin/reboot the system.
This should fix it

EdXian, 2017-11-12:
Hi @ev-mp , 
The problem has been solved . 
Thanks

closed 2017-11-12 by EdXian

tony497, 2018-03-20:
First of all, thank you for your messages it helped me a lot to build intelRealSense.
So i was able to install successfully the following lines
```
mkdir build && cd build
cmake .. -DBUILD_EXAMPLES=true
make
sudo make install
```
But now when i go to usr/local/bin and try to run cpp-something i get 
`There are 0 connected RealSense devices.`
even if my realsense R200 is connected, do you have any idea ?

EdXian, 2018-03-20:
hi @tony497,
If  you use R200 camera,  you should use [legacy](https://github.com/IntelRealSense/librealsense/tree/legacy#compatible-devices) branch.

tony497, 2018-03-20:
hi @EdXian ,
Yes it was already the installed branch.
I just fixed the problem, of curse very stupid has always, i changed my cable and used it on the proper usb port and not on the docker station. 

never the less, thanks for your help

AndreaIncertiDelmonte, 2018-05-22:
Hi @EdXian and @ev-mp,
Regarding 

> check whether the file is registered with Ubuntu by which librealsense.so - if not present then run ldconfig to refresh the so files cache, or relogin/reboot the system.
This should fix it

We need also to make sure that `/usr/bin/lib/` is inside the `LD_LIBRARY_PATH` env variable.
To this end is possible to use:
`export LD_LIBRARY_PATH=/usr/local/lib/${LD_LIBRARY_PATH:+:${LD_LIBRARY_PATH}}`

aranganath, 2018-07-13:
I tried all of this. But I ran into a lot of issues.

The thing that worked for me was sudo apt-get remove ros-<distro>-librealsense ros-<distro>-realsense-camera

Then

sudo apt-get install ros-<distro>-librealsense ros-<distro>-relsense-camera