realsenseai/librealsense #815: Incorrect PNG file format for Viewer?

MichaelHanssonPH, 2017-11-13:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0  | 
| Operating System & Version | Windows 10 |
| Kernel Version (Linux Only)  |   |

Does anyone else have this problem?  The 2D PNG files written by the Intel RealSense Viewer (v2.8.1) appear to have an incorrect format. The cam works fine for me and both 2D and 3D images look correct in the viewer. The 3D .PLY files also appear correct. But I cannot get the PNG files generated (sample attached) to open correctly in any program. ImageJ claims that they are RGB images with all R/G/B values being equal and values ranging only from 228-255 (i.e. 8-bit and of limited range). A low-level PNG viewer such as TweakPNG says that the image has 8 bits per sample with grayscale+alpha data. Gimp2 also thinks there's an alpha layer. Nothing I've tried loads a 16-bit grayscale image, as expected. Using TweakPNG I edited the PNG image header forcing the data to load as a 16-bit grayscale image, but the image is still grossly incorrect (values along what is a flat surface swing from 1,000-60,000).

I am a user and a rusty programmer, and have not looked at the source code for the Viewer.

[image]

ev-mp, 2017-11-14:
Hello @MichaelHanssonPH ,
The Depth image PNG format is as follows - [Depth Width (pixels), Depth Height (pixels) , Compression (2 bytes), Stride (Width*2)].
Note that in the 'realsense-viewer' the depth data is post-processed to generate a colorized image with palettes like the Matlab's "Jet" and "Classic" to make it "human-readable".
The PNG snapshot, however,  preserves the original raw data (Depth, 16bit per-pixel,unsigned) that can be directly imported into 3rd party tools (e.g Matlab), and therefore is best suited for machine processing.

In case you need the colored snapshot of the depth data you can use  'rs-save-to-disk' demo that does precisely this.

MichaelHanssonPH, 2017-11-14:
Hi @ev-mp ,
Sorry if I wasn't clear. I understand that the depth data in the viewer is "artificially" colored. I just need to access the depth data using 3rd party tools for analysis to, for example, evaluate potential algorithm steps for edge detection, filtering, etc. I would have thought that the images could be opened directly (as 16-bit grayscale images) in software such as ImageJ but it seems they do not. I tried Python+OpenCV and I have the same problem. I have not tried Matlab.

ev-mp, 2017-11-14:
Hi @MichaelHanssonPH ,
When looking into the polarized snapshot attached, it could actually be an issue of endianness as it looks like that the upper and the lower bytes data are swapped.
The depth raw data that is stored arrives in LSB order, while according to this [spec](https://www.w3.org/TR/REC-png-961001) PNG expects the 16bit pixel to be MSB-aligned
> ...16-bit samples are stored in network byte order (MSB first).

We'll look further into this as it seem to be a software issue. Meanwhile you can try to reverse the endianness or just treat the raw data as a binary stream rather than PNG image.

MichaelHanssonPH, 2017-11-14:
Hi @ev-mp ,
Thank you. I was beginning to suspect the same thing. Reading the raw data might be challenging, considering the compression, but I suppose I can read it as a PNG and then swap the bytes over. I will try it.

MichaelHanssonPH, 2017-11-14:
Hi again @ev-mp ,

I can confirm that the IHDR chunk of the PNG file is wrong, indicating an 8-bit grayscale+Alpha sample instead of a 16-bit grayscale sample. I can also confirm that the MSB and LSB bytes for each value is incorrectly ordered.

If I edit the IHDR with TweakPNG and then swap the bytes over (in my case with .byteswap() in Python), the images now appear correct.

freemanlo, 2017-11-15:
I also found the realsense viewer saves the depth image as 32bit depth instead of 16bit. Use the opencv imwrite to save the depth image as "CV_16U" without the issue.

dorodnic, 2017-12-11:
For [version 2.8.3](https://github.com/IntelRealSense/librealsense/releases/tag/v2.8.3) we went back to saving depth snapshots as 24-bit RGB PNG files. Hopefully this is more convenient for most users.

closed 2017-12-11 by dorodnic

apoorva2398, 2018-07-24:
Hi, I also have an issue with the endianness of depth raw data. Reading it as little endian gives the correct depth values for each pixel but doesn't display correctly, and vice versa for big endian. 

Little endian gives the correct depth value (499 mm) but displays the image incorrectly (with colormap jet):
[image]

Big endian displays the image correctly but gives the wrong depth value (6221 mm >> 499 mm!):
[image]

RealSense Viewer snapshot:
[image]

cygerts, 2019-02-19:
So how can you actually read the .raw file in python?