realsenseai/librealsense #825: Are there Findlibrealsense.cmake provided?

csuwp, 2017-11-16:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |

dorodnic, 2017-11-22:
Hi @csuwp 
That's a great question, at the moment we do not provide one, but we should and will.
@zivsha has previously provided a sample for CMake integration [here](https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started). This is not quite what you want but it might help you get started.

csuwp, 2017-11-28:
Thanks for reply, but I am using the this lib in ubuntu16.04.The cmake you provide seems don't work

sogartar, 2017-12-07:
librealsense2 installs
```
/usr/local/lib/cmake/realsense2/realsense2Targets.cmake
/usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake
/usr/local/lib/cmake/realsense2/realsense2Config.cmake
```
In your CMake you can have
```
set(LIBREALSENSE_INSTALLATION_DIR "" CACHE PATH "librealsense installation directory")
if(LIBREALSENSE_INSTALLATION_DIR STREQUAL "")
    find_package(realsense2 REQUIRED)
else()
    find_package(realsense2 REQUIRED PATHS ${LIBREALSENSE_INSTALLATION_DIR} NO_DEFAULT_PATH)
endif()

...

target_link_libraries(my-target PRIVATE realsense2::realsense2)
```

For 1.0 I am not sure if you can do that.

lazysquid, 2018-03-19:
I have one question @sogartar 
I followed this [document](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) to install librealsense2 and `find_package(realsense2)` works smoothly
However when I check 
```
/usr/local/lib/cmake/realsense2/realsense2Targets.cmake
/usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake
/usr/local/lib/cmake/realsense2/realsense2Config.cmake
```
There are no such things
Also I searched those three files from root to bottom but I failed to find them.
Then how my cmake can find `realsense2`?

dpkg -l showed realsense libraries however I couldn't find any connection between `find_package` function and dpkg

closed 2018-05-14 by RealSense-Customer-Engineering

gayanbrahmanage, 2018-10-11:
Use following template for realsense intergartion in CMake
`cmake_minimum_required(VERSION 2.8)
project(project_name)

set(CMAKE_EXPORT_COMPILE_COMMANDS 1)

include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX11)
    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c11")
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
elseif(COMPILER_SUPPORTS_CXX0X)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
endif()

find_package(realsense2 REQUIRED )

add_executable( target_name src/source.cpp )
include_directories(include ${realsense_INCLUDE_DIR})
target_include_directories(target_name PRIVATE ${realsense_INCLUDE_DIR})
target_link_libraries(rs-align ${realsense2_LIBRARY})`