realsenseai/librealsense #828: python-tutorial-1-depth.py fails when running under Python3

mkhansen-intel, 2017-11-16:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04  |
| Kernel Version (Linux Only)  |   4.4.0.38 generic |

There seems to be some issues with the python tutorial. 
First, when running using Python 2, it can't find the pyrealsense2 library, even when it exists in the PYTHONPATH variable
*****************
mkhansen@mkhansen-sky1:~/librealsense/build/wrappers/python$ python ~/librealsense/wrappers/python/python-tutorial-1-depth.py 
Traceback (most recent call last):
  File "/home/mkhansen/librealsense/wrappers/python/python-tutorial-1-depth.py", line 9, in <module>
    import pyrealsense2 as rs
ImportError: No module named pyrealsense2
mkhansen@mkhansen-sky1:~/librealsense/build/wrappers/python$ ls
CMakeFiles
cmake_install.cmake
Makefile
pybackend2.cpython-35m-x86_64-linux-gnu.so
pybackend2.cpython-35m-x86_64-linux-gnu.so.2
pybackend2.cpython-35m-x86_64-linux-gnu.so.2.8.1
pyrealsense2.cpython-35m-x86_64-linux-gnu.so
pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2
pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2.8.1
third_party
***********************
Second, I worked around this by running Python3, but ran into this error:
mkhansen@mkhansen-sky1:~/librealsense/build/wrappers/python$ python3 ~/librealsense/wrappers/python/python-tutorial-1-depth.py 
name 'xrange' is not defined

That was fixed by changing 'xrange' to 'range', however, then I get this error:
mkhansen@mkhansen-sky1:~/librealsense/build/wrappers/python$ python3 ~/librealsense/wrappers/python/python-tutorial-1-depth.py 
string indices must be integers

I suspect this is coming from this line of code:
                for c in coverage:
                    line += " .:nhBXWW"[c/25]

Admittedly, I don't know why [c/25] is the index being used in this case, so this is where I'm stuck. Is this the real issue? Has this tutorial been known to work before?

zivsha, 2017-11-17:
Hi,
The Python examples are partially in Python2 and the rest for Python3. We really should split them to make this clearer or have a cross version examples. If you feel like contributing we would appreciate a PR with such a change.

If you build `pyrealsense` against Python2 you will only be able to `import pyrealsense` when running using Python2 , and the same goes for Python3.
Please verify with which version of python you are compiling the library and use that version.

zivsha, 2017-11-17:
You can workaround the `c/25` using `c//25`

mkhansen-intel, 2017-11-17:
OK, I fixed the issues and submitted PR #830

closed 2017-11-29 by dorodnic

ROY-SHIN, 2020-04-10:
xrange function is python2.x
so you can use range function in python3.X