realsenseai/librealsense #831: libusb_get_parent

ChampionMas, 2017-11-18:
| Required Info  |   |
|---|---|
| Camera Model | R200  | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  |  Linux version 4.4.0-98-generic (buildd@lcy01-03) (gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4) )  |

I fellow the instructions of installation;
when I executed `realsense-viewer`,
it return below:
```
realsense-viewer: symbol lookup error: /usr/local/lib/librealsense2.so.2: undefined symbol: libusb_get_parent
``` 
I have reinstalled the libusb-1.0-0-dev for several times, but still useless.
What should I do?

icarpis, 2017-11-20:
Hi @ChampionMas, please try to do what they offering at this [thread](https://communities.intel.com/thread/117119).
Generally, the `ldd` command on Linux is a tool that prints the dependencies of a binary file. You can try to use it on the realsense-viewer tool (`$ ldd realsense-viewer`) to figure out where it expecting to find the libusb file.

freemanlo, 2017-11-22:
Hi @ChampionMas 

Please use below legacy librealsense1 build instead of librealsense2 which is for D400 series and SR300.
https://github.com/IntelRealSense/librealsense/tree/legacy

ChampionMas, 2017-11-22:
Great Respects for @icarpis and @freemanlo ! Thank you !

I tried `legacy` and fellow the instructions of installation and it did work.
In addition, I found that when compile, we had better add `-lusb-1.0` for libusb-1.0.18 or later version. Just like: (for mono source file)
```
g++ source.cpp -o target -lusb-1.0 -lrealsense `pkg-config --cflags --libs opencv`
```
or it will show errors:
```
/usr/local/lib/librealsense.so: undefined reference to `libusb_strerror'
/usr/local/lib/librealsense.so: undefined reference to `libusb_get_port_numbers'
/usr/local/lib/librealsense.so: undefined reference to `libusb_get_parent'
/usr/local/lib/librealsense.so: undefined reference to `libusb_error_name'
collect2: error: ld returned 1 exit status
```
Hope that helps others !

closed 2017-11-27 by dorodnic

darknight-007, 2018-08-17:
For me, this error appeared, after initially things working fine. It could be as a result of me installing MatrixVision mvAcquire SDK. Just wanted to point out that in my case this was not due to incorrect version of librealsense as things were working fine.