realsenseai/librealsense #835: ZR 300 on Windows

NikoBele1, 2017-11-20:
Hello,
did someone manage to use the ZR300  on Windows? Especially for skeleton tracking / gesture detection.

Regards
Niko

freemanlo, 2017-11-22:
Hi @NikoBele1 
Do you still use RealSense Windows SDK? If so, the DCM driver is not available for ZR300.

dorodnic, 2017-12-01:
[librealsense1](https://github.com/IntelRealSense/librealsense/releases/tag/v1.12.1) should work with ZR300 on Windows, however, rest of the software stack ([Linux SDK](https://github.com/IntelRealSense/realsense_sdk_zr300) and the CV middlewares) were published for Linux only, with strong emphasis on integration with Intel Joule platform.

closed 2017-12-01 by dorodnic

linuxchunk, 2025-02-11:
Is there someone who's working on the ZR300 now? it can be used in ros2?

MartyG-RealSense, 2025-02-11:
Hi @linuxchunk  The RealSense ZR300 was a stereo depth camera that was released in-between the R200 and D435 models.  It has been retired and unsupported for many years, before support for ROS2 with RealSense cameras was introduced.

ZR300 was supported in the RealSense ROS1 wrapper, from version 1.6.0 and was not referred to again after version 1.8.1.

https://github.com/IntelRealSense/realsense-ros/releases/tag/1.6.0
https://github.com/IntelRealSense/realsense-ros/releases/tag/1.8.1

These wrappers required the use of the 'legacy librealsense' SDK 1.12.1.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

If you wanted a modern-day equivalent of the ZR300 camera that is supported in the RealSense ROS2 wrapper then a RealSense 400 Series camera model such as D435 will be suitable.