realsenseai/librealsense #836: Only RGB camera works on Windows 10 Creators Update. Depth camera crashes

danielw256, 2017-11-20:
I'm able to install the Windows drivers and this library on my Dell laptop.
When I run realsense-viewer.exe, I can start the RGB camera and get a video feed.
When I start the coded-light depth sensor, a child window open and closes, the camera sources close, the usb plugin sound plays, and the camera sources reopen.
In my Device Manager, under the Camera node, I have the Intel SR300 Depth node and the Intel SR300 RGB node.
Under the Imaging devices node, I have the Intel SR300 Virtual Driver node.
Under the USB devices node, I have the Intel Camera SR300 node.
(I have another laptop with the Intel SR300 Depth node and the Intel SR300 RGB node under the  Imaging devices node, and it still crashes.)

I know the Windows DCM drivers do not work in Creators Update, and Intel is no longer updating them.
But is this library supposed to work in Creators Update, or is a fix coming?

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0 | 
| Operating System & Version | Win10 Pro version 1709 build 16299.19 |

freemanlo, 2017-11-22:
@danielw256 
which DCM version do you use? 
Is the SR300 your have a Creative Senz3D peripheral?
Does this issue happen after RS2 to RS3 upgrade?

danielw256, 2017-11-22:
From Device Manager:

RGB driver
Microsoft
6/21/2006
10.0.16299.15

Depth driver
Microsoft
6/21/2006
10.0.16299.15

Virtual Driver
Intel
6/8/2016
12.11.38.553

Is there any other way to get the DCM version?
I bought the SR300 from Intel this November. There's no mention of Creative Senz3D.
I'm using RealSense SDK 2.0. What's RS3?

freemanlo, 2017-12-04:
Hi @danielw256 
My PC Win10 version is 10.0.16299.15 Build 16299. DCM version = 3.3.27.5718 (you can find the version info in "App & features"). It looks your virtual driver and SR300 FW version are the same with mine. 

For the device showing in the device manager:

1. The Intel SR300 Depth node and Virtual driver under the Imaging devices node. 
2. The Intel SR300 RGB node under the Cameras node. 
3. The Intel Camera SR300 node under the USB devices node.

However, I use the RealSense Viewer in librealsense 2.8.2 to view the color and depth streams without any issues as you can see the attached picture. Does the depth crash issue happen after your Win10 OS upgrade?

[image]

danielw256, 2017-12-04:
@freemanlo 
Thanks for your help.

I finally got the SR300 to work. 
I plugged the SR300 into a powered USB 3 hub
I followed the instructions in the following webpage
How to get SR300 working on Windows 10 Build 1703 - Including SDK Apps such as Segmentation (Updated 5/17/2017) 
I could not get the instructions to work as written. Mainly, due to no action center or toast notification.
I believe 2 main things got it to work.
(1) The Intel depth driver replaced a Microsoft depth driver
(2) The Intel virtual driver loaded.

Daniel

closed 2017-12-08 by jlgarcia75