realsenseai/librealsense #837: D415/D435 RGB image is not rectified at rsutil.h

TetsuriSonoda, 2017-11-20:
rs-align example with D415 and D435 is causing offset between color and depth.

On librealsense, D435 and D415 will return its distortion model as RS2_DISTORTION_BROWN_CONRADY.
But this distortion model is not handled at rs2_project_point_to_pixel in rsutil.h 
(Only handled RS2_DISTORTION_MODIFIED_BROWN_CONRADY and RS2_DISTORTION_FTHETA).
I think it needs to be undistorted at there.

| Required Info  |   |
|---|---|
| Camera Model | D415 and D435 | 
| Firmware Version |  05.08.05.00 | 
| Operating System & Version | Windows8  |
| Kernel Version (Linux Only)  |   |

cynosure4sure, 2017-11-25:
Are D415 and D435 available for purchase?

halton, 2017-11-27:
https://click.intel.com/intelr-realsensetm-depth-camera-d415.html, out-of-stock

danielw256, 2017-11-27:
I will need rectified images and depth, too. Please fix.

ev-mp, 2017-11-27:
Hi @TetsuriSonoda , The depth images received from the device are already rectified with RS2_DISTORTION_BROWN_CONRADY model. Therefore `rs2_project_point_to_pixel`  will just pass them through directly without any additional rectifications.

TetsuriSonoda, 2017-11-27:
Hi, @ev-mp , this issue not about depth image, but RGB image side.
For aligned depth image to RGB, we need "rectified RGB" image with proper undistortion.
But such image is not provided correctly by librealsense2 now.

closed 2017-12-03 by dorodnic

karlita101, 2020-10-26:
> Hi, @ev-mp , this issue not about depth image, but RGB image side.
> For aligned depth image to RGB, we need "rectified RGB" image with proper undistortion.
> But such image is not provided correctly by librealsense2 now.

Would it be  possible to get further clarification on why the RGB are not undistorted ( my understanding is that if the coefficients are set 2 zero, no undistortion is taking place).