realsenseai/librealsense #839: No 3D pointcloud is displayed in realsense-viewer

drewm1980, 2017-11-21:
| Required Info  |   |
|---|---|
| Camera Model | SR300 / RS410 | 
| Firmware Version | 3.17.0.0 for the SR300  | 
| Operating System & Version | Ubuntu 16.04.3 LTS  |
| Kernel Version (Linux Only)  | 4.10.0-38-generic #42~16.04.1-Ubuntu  |

I see the wireframe for the camera frustrum, but I do no see any 3D pointcloud data in realsense-viewer. I tried both the ubuntu package, and reproduced the behavior building from source. The camera is on, and I can see the 3D depth stream. I reproduced it with cameras of two different models, an SR300 and a RS410 engineering sample. I performed a hardware reset on both cameras to make sure it wasn't an issue with the settings.

I believe my OpenGL environment is OK; glxgears works, and as I specified above, the wireframe renders and can be manipulated with the mouse; it is just the data that is missing. glxinfo shows that my gpu (NVIDIA Quadro P600) is indeed being used, rather than nouveau:

$ glxinfo | head
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: NVIDIA Corporation
server glx version string: 1.4
server glx extensions:

halton, 2017-11-22:
What version of librealsense you're using?

drewm1980, 2017-11-22:
$ locate "librealsense2.so"
/home/awagner/src/realsense/librealsense/librealsense2.so
/home/awagner/src/realsense/librealsense/librealsense2.so.2
/home/awagner/src/realsense/librealsense/librealsense2.so.2.8.1
/usr/lib/x86_64-linux-gnu/librealsense2.so
/usr/lib/x86_64-linux-gnu/librealsense2.so.2
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.8.1

I double checked that that's the version my realsense viewer is using via strace:
$ strace -ofoo realsense-viewer
$ grep open foo | grep "realsense2.so" | grep -v ENOENT
open("/usr/lib/x86_64-linux-gnu/librealsense2.so.2", O_RDONLY|O_CLOEXEC) = 3
open("/usr/lib/x86_64-linux-gnu/librealsense2.so.2.8.1", O_RDONLY) = 4
open("/usr/lib/x86_64-linux-gnu/librealsense2.so.2.8.1", O_RDONLY) = 4

drewm1980, 2017-11-22:
If I compile from source, it's correctly linking against the librealsense that was built:

awagner@rv-awagner:/home/awagner/src/realsense/librealsense/tools/realsense-viewer
$ strace -ofoo realsense-viewer 

$ grep open foo | grep "realsense2.so" | grep -v ENOENT
open("/home/awagner/src/realsense/librealsense/librealsense2.so.2", O_RDONLY|O_CLOEXEC) = 3
open("/home/awagner/src/realsense/librealsense/librealsense2.so.2.8.1", O_RDONLY) = 4
open("/home/awagner/src/realsense/librealsense/librealsense2.so.2.8.1", O_RDONLY) = 4

I compiled the development branch at:
commit 58a5c437a4284fd351cecca592f1189188aff82c
Merge: b651eed eb5ff05
Author: Halton Huo <halton.huo@intel.com>
Date:   Mon Nov 20 11:35:15 2017 +0800

Is it possible the code is relying on some OpenGL extension not supported by my Quadro? It has to be something that would affect the display of the data, but not the wireframe... Maybe gllines are working but not whatever you're using for the points...

dorodnic, 2017-11-22:
Hi @drewm1980 
The viewer is using only very basic OpenGL commands, no extensions no shaders. 
I suspect this might be a software bug. In latest releases we seem to not render the depth when no texture is provided. Can you retry but enable both depth and IR streams?

dorodnic, 2017-12-11:
This should be resolved in [2.8.3](https://github.com/IntelRealSense/librealsense/releases/tag/v2.8.3). Please give it a try and reopen the issue if you see the problem.

closed 2017-12-11 by dorodnic