realsenseai/librealsense #840: SR300 is only working on alternate runs

syedharoonalam, 2017-11-21:
| Required Info  |   |
|---|---|
| Camera Model |  SR300  | 
| Firmware Version | 3.21  | 
| Operating System & Version | Jetson TX2-L4T OS 28.1  |
| Kernel Version (Linux Only)  | 4.4.38  |

On Realsense sdk2 2.8.1, SR300 only works on alternate runs. First time camera extracts data, but second time application exits with error: 

**rs_error was raised when calling rs2_pipeline_wait_for_frames(pipe:0x437740): Frame didn't arrived within 5000** 
OR
**RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x766ad0): Frame didn't arrived within 5000**

The error is same in sdk's all examples (C and C++). I made sure that pipe.stop() in C++, and code in _//Stop the pipeline streaming //Release resources_ in C gets called. 

Please advice.

dorodnic, 2017-11-22:
Hi @syedharoonalam 
Jetson TX2 platform does not officially support RealSense, and there are known issues with this setup. 
However, if you could provide librealsense logs and dmesg logs during the problematic runs, this can help us isolate the problem. ([how to enable librealsense logs](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-how-do-i-enable-librealsense-logs))

syedharoonalam, 2017-11-22:
@dorodnic  Here is the dmesg logs after librealsense crashes: 

`[41538.853326] Call trace:
[41538.855768] [<ffffffc00077a30c>] v4l_enum_fmt+0x1074/0x1400
[41538.861329] [<ffffffc00077aea4>] __video_do_ioctl+0x25c/0x2c8
[41538.867065] [<ffffffc00077a92c>] video_usercopy+0x294/0x590
[41538.872627] [<ffffffc00077ac3c>] video_ioctl2+0x14/0x20
[41538.877843] [<ffffffc000775d18>] v4l2_ioctl+0x110/0x118
[41538.883060] [<ffffffc0001dc52c>] do_vfs_ioctl+0x33c/0x5f0
[41538.888448] [<ffffffc0001dc86c>] SyS_ioctl+0x8c/0xa0
[41538.893405] [<ffffffc000084ff0>] el0_svc_naked+0x24/0x28
[41538.919331] uvcvideo: Non-zero status (-71) in video completion handler.
[41538.937026] uvcvideo: Non-zero status (-71) in video completion handler.
[41547.748287] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 5
[41547.755408] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_DEC_SSPI_CLOCK request`

And here is the log after successfully running librealsense: 

`[41743.798443] Call trace:
[41743.800884] [<ffffffc00077a30c>] v4l_enum_fmt+0x1074/0x1400
[41743.806446] [<ffffffc00077aea4>] __video_do_ioctl+0x25c/0x2c8
[41743.812180] [<ffffffc00077a92c>] video_usercopy+0x294/0x590
[41743.817740] [<ffffffc00077ac3c>] video_ioctl2+0x14/0x20
[41743.822953] [<ffffffc000775d18>] v4l2_ioctl+0x110/0x118
[41743.828170] [<ffffffc0001dc52c>] do_vfs_ioctl+0x33c/0x5f0
[41743.833557] [<ffffffc0001dc86c>] SyS_ioctl+0x8c/0xa0
[41743.838512] [<ffffffc000084ff0>] el0_svc_naked+0x24/0x28
[41753.815436] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 5
[41753.822500] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_DEC_SSPI_CLOCK request
`
I can see after the crash, dmesg mentions few lines in the end as: uvcvideo: Non-zero status (-71) in video completion handler. Do you think this should be the problem, and any suggestion to fix this error?

Also, can you please let me know me where should I find librealsense logs file ? I am not able to get logs from the command you mentioned.

lesreaper, 2017-11-23:
@syedharoonalam I replied to your info on the JetsonHack site. How did you ever get the Depth image to stop flickering and the IR data to appear on the TX2? The INZI problem is something no one seems to be able to solve!

syedharoonalam, 2017-11-24:
@lesreaper you need to apply a patch for INZI support and rebuild the UVC module. Please follow: https://github.com/syedharoonalam/buildJetsonTX2Kernel

lesreaper, 2017-11-25:
This is awesome @syedharoonalam ! Great work. Very clear instructions, and only one thing that got glossed over a little in the instructions.

When you said ["follow the patch by realsense"](https://cl.ly/0G2P3p151H0s), which patch are you referring to for the TX2?

[Recommended Patch by Intel 1](https://cl.ly/3J3H263M3c2f)
[All Available Scripts](https://cl.ly/251C2p31092U)

lesreaper, 2017-12-03:
@syedharoonalam Thanks for the help, I got it all working to some extent. You and I seem to be running into a lot of the same issues.

The USB3 connection b/w the Realsense and Jetson fails a lot.

Were you ever able to get your problem here to work with the Jetson?

Laxen, 2018-01-08:
@icarpis I'm having the same problem with the SR300, it only works every other time. The problem can easily be replicated using the realsense-viewer program, when enabling/disabling the depth/color stream it will show a white screen with "No Frames Received!" text every other time. After a while a notification will come up saying "Frames didn't arrive within 5 seconds". 

This can be solved in code by simply restarting the pipeline using stop() and start(). However, there is a problem when the two streams become "unsynchronized" such that the depth stream receives no frames while the color stream works like it should, and vice versa. When this happens you have to restart only one of the streams to get them "synchronized" again. 

[image]
This image shows the problem when the depth stream works but the color stream doesn't receive any frames. In order to solve this the color stream has to be restarted manually (on/off button). 

Is there some better way to solve this?

Laxen, 2018-01-22:
I just got a D435 and I have the same problem. This seems to be an issue in librealsense and not hardware related. I'm running librealsense 2.9.0.

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @Laxen

1. For the issue "it only works every other time", please use the latest L4T release (28.2), not 28.1. You can refer to below to get the latest BSP 28.2. And then, apply the realsense camera patches. This should resolve the issue.
https://github.com/aivero/buildJetsonTX2Kernel/blob/fix/jetson_tx2/scripts/getKernelSources_noGUI.sh
2. For 2nd issue about no frames received for color steam, this is probably due to the overall processing bandwidth. You can consider to change the resolution from 1920x1080 to 960x540 or 640x480. "No Frames Received" should not happen.

RealSense-Customer-Engineering, 2018-03-09:
[Realsense Customer Engineering Team Comment] 
Hi @syedharoonalam

Still need any help?

syedharoonalam, 2018-03-09:
Thanks, @RealSense-Customer-Engineering. I will check out the latest L4T and get back if found any issues. Thanks for fixing it for us :)

RealSense-Customer-Engineering, 2018-03-25:
[Realsense Customer Engineering Team Comment] 
If you still has the issue, please refer to below patch for the latest BSP (L4T 28.2).
Copy the patch file to ./scripts folder and run the patch: ./scripts/realsense-ubuntu-xenial-jetson-tx2.sh to patch the uvc modules.
https://github.com/freemanlo/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial-jetson-tx2.sh

closed 2018-03-25 by RealSense-Customer-Engineering