realsenseai/librealsense #844: no realsense connected when run intel realsense viewer

xiaochongs, 2017-11-22:
| Required Info  |   |
|---|---|
| Camera Model |  / F200 
| Firmware Version |   | 
| Operating System & Version |   |Ubuntu 16.04 64 bit
| Kernel Version (Linux Only)  |   |4.4.0-78-generic

I follow the installation guide, but in the last, it is still not working.
when run intel viewer, no realsense connected.
when run ./rs-capture
rs-capture: /tmp/glfw/src/window.c:774: glfwSetWindowUserPointer: Assertion `window != NULL' failed.

the dmesg log:
[Wed Nov 22 02:18:24 2017] usb 4-4: Product: Intel(R) RealSense(TM) 3D Camera (Front F200)
[Wed Nov 22 02:18:24 2017] usb 4-4: Manufacturer: Intel(R) RealSense(TM) 3D Camera (Front F200)
[Wed Nov 22 02:18:24 2017] usb 4-4: SerialNumber: 044150126005
[Wed Nov 22 02:18:24 2017] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
[Wed Nov 22 02:18:24 2017] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0/input/input19
[Wed Nov 22 02:18:24 2017] uvcvideo: Unknown video format 5a564e49-2d90-4a58-920b-773f1f2c556b
[Wed Nov 22 02:18:24 2017] uvcvideo: Unknown video format 495a4e49-1a66-a242-9065-d01814a8ef8a
[Wed Nov 22 02:18:24 2017] uvcvideo: Unknown video format 52564e49-2d90-4a58-920b-773f1f2c556b
[Wed Nov 22 02:18:24 2017] uvcvideo: Unknown video format 49524e49-2d90-4a58-920b-773f1f2c556b
[Wed Nov 22 02:18:24 2017] uvcvideo: Unknown video format 49564e49-57db-5e49-8e3f-f479532b946f
[Wed Nov 22 02:18:24 2017] uvcvideo: Unknown video format 494c4552-1314-f943-a75a-ee6bbf012e23
[Wed Nov 22 02:18:24 2017] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
[Wed Nov 22 02:18:24 2017] uvcvideo: Unable to create debugfs 4-5 directory.
[Wed Nov 22 02:18:24 2017] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.2/input/input20
[Wed Nov 22 02:18:25 2017] xhci_hcd 0000:00:14.0: Cannot set link state.
[Wed Nov 22 02:18:25 2017] usb usb4-port4: cannot disable (err = -32)
[Wed Nov 22 02:18:25 2017] usb 4-4: USB disconnect, device number 5
[Wed Nov 22 02:18:28 2017] usb 4-1: new SuperSpeed USB device number 6 using xhci_hcd
[Wed Nov 22 02:18:28 2017] usb 4-1: New USB device found, idVendor=8086, idProduct=0a66
[Wed Nov 22 02:18:28 2017] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Wed Nov 22 02:18:28 2017] usb 4-1: Product: Intel(R) RealSense(TM) 3D Camera (Front F200)
[Wed Nov 22 02:18:28 2017] usb 4-1: Manufacturer: Intel(R) RealSense(TM) 3D Camera (Front F200)
[Wed Nov 22 02:18:28 2017] usb 4-1: SerialNumber: 044150126005
[Wed Nov 22 02:18:28 2017] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
[Wed Nov 22 02:18:28 2017] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/input/input21
[Wed Nov 22 02:18:28 2017] uvcvideo: Unknown video format 5a564e49-2d90-4a58-920b-773f1f2c556b
[Wed Nov 22 02:18:28 2017] uvcvideo: Unknown video format 495a4e49-1a66-a242-9065-d01814a8ef8a
[Wed Nov 22 02:18:28 2017] uvcvideo: Unknown video format 52564e49-2d90-4a58-920b-773f1f2c556b
[Wed Nov 22 02:18:28 2017] uvcvideo: Unknown video format 49524e49-2d90-4a58-920b-773f1f2c556b
[Wed Nov 22 02:18:28 2017] uvcvideo: Unknown video format 49564e49-57db-5e49-8e3f-f479532b946f
[Wed Nov 22 02:18:28 2017] uvcvideo: Unknown video format 494c4552-1314-f943-a75a-ee6bbf012e23
[Wed Nov 22 02:18:28 2017] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
[Wed Nov 22 02:18:28 2017] uvcvideo: Unable to create debugfs 4-6 directory.
[Wed Nov 22 02:18:28 2017] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.2/input/input22

modinfo uvcvideo log:
filename:       /lib/modules/4.4.0-78-generic/kernel/drivers/media/usb/uvc/uvcvideo.ko
version:        1.1.1
license:        GPL
description:    USB Video Class driver
author:         Laurent Pinchart <laurent.pinchart@ideasonboard.com>
srcversion:     E8B702DAAE2663578293C90
alias:          usb:v*p*d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v2833p0201d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1C4Fp3000d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1B3Bp2951d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d00*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d01[0-1]*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v19ABp1000d012[0-6]dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v199Ep8102d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v18ECp3290d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v18ECp3288d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v18ECp3188d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v18CDpCAFEd*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v1871p0516d*dc*dsc*dp*icFFisc01ip00in*
alias:          usb:v1871p0306d*dc*dsc*dp*ic0Eisc01ip00in*
alias:          usb:v17EFp480Bd*dc*dsc*dp*ic0Eisc01ip00in*
[... 84 lines cut ...]
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 0.500s (2.000 fps)

Thanks.

dorodnic, 2017-11-22:
At the moment the Viewer does not support F200 devices. As an alternative, you can check out the [legacy software stack](https://github.com/IntelRealSense/librealsense/tree/legacy), offering much of the same functionality.

xiaochongs, 2017-11-23:
thank you very much.
It works.

closed 2017-11-23 by dorodnic