realsenseai/librealsense #846: missing media fold while degrade kernel to match patch

zhuzunjie17, 2017-11-23:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | 1.12.1  | 
| Operating System & Version | ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.10.0-40 degraded to 4.4.0 |

the problem I've met is while I degrading kernel to 4.4.0, I can't find the media fold in **/lib/modules/4.4.0-X-generic/kernel/drivers/**  , where the uvcvideo.ko should be copy into (**/media/usb/uvc/**). And once I `mkdir `the fold by myself and `cp`the uvcvideo.ko follow  **patch-uvcvideo-16.04.simple.sh**, then another question come to me while `sudo modprobe uvcvideo `, 

> modprobe: FATAL: Module uvcvideo not found in directory /lib/modules/4.4. 0-041214-generic

so, i'm wondering what i should do to degrade kernel with media fold?  or does it have patch for 4.10.0 ?

dorodnic, 2017-11-23:
ZR300 device does not require Kernel patches for 4.10 kernel (and 4.4 since 4.4.0.56 if I recall correctly)
After setting-up the [udev-rules](https://github.com/IntelRealSense/librealsense/blob/legacy/doc/installation.md#video4linux-backend-preparation) and building the [library](https://github.com/IntelRealSense/librealsense/tree/legacy) from source, you should be able to stream Depth, IR, Color, Fish-eye and Motion data using `cpp-config-ui` utility.

dorodnic, 2017-12-01:
Please re-open the issue if you have further questions

closed 2017-12-01 by dorodnic