realsenseai/librealsense #847: Floor Removal

dolhasz, 2017-11-23:
| Required Info  |   |
|---|---|
| Camera Model |SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |

Hi there,

Are there any plans (or implementations) of floor removal in librealsense? Seems it would be a really useful feature. 

Regards,

Alan

dorodnic, 2017-11-24:
Hi @dolhasz 
At the moment I am not aware any plans for such feature. Could you elaborate what use-case you have in mind? We will be happy to look into it.

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @dolhasz ,

Do you still need help with this issue?

 Regards

closed 2018-03-09 by RealSense-Customer-Engineering

noumanriazkhan, 2018-10-30:
I am also looking for floor segmentation or detecting if an object (I know of in 2d) touches the ground. How is possible in python?