realsenseai/librealsense #848: make error

sunnyclaracn, 2017-11-24:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version |   | ubuntu 16.04
| Kernel Version (Linux Only)  |   |

When make the package, below error is given. Any one can help? 
=======================================
 error: could not convert ‘sensor->rs2_sensor::parent’ from ‘const rs2_device’ to ‘std::shared_ptr<librealsense::context>’
     return new rs2_device { sensor->parent };
=====================================

dorodnic, 2017-11-24:
Hi @sunnyclaracn 
What OS and compiler version are you using? `librealsense` requires GCC5 or Visual Studio 15. 
We also provide pre-compiled binaries for [Windows](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_windows.md) and [Ubuntu 16.04](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) users if you prefer not to build from source.

mjfoo21, 2017-11-30:
Hello @dorodnic 

I got this error as well. Even though I installed GCC5 (version 5.4.1), I still got the following:

```
[ 57%] Linking CXX executable rs-capture
../../librealsense2.so.2.8.2: undefined reference to `librealsense::context::add_device(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../librealsense2.so.2.8.2: undefined reference to `el::Loggers::reconfigureLogger(std::string const&, el::Configurations const&)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::datetime_string()'
../../librealsense2.so.2.8.2: undefined reference to `rosbag::Bag::toHeaderString[abi:cxx11](ros::Time const*) const'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::device_hub::wait_for_device(std::chrono::duration<long, std::ratio<1l, 1000l> > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../librealsense2.so.2.8.2: undefined reference to `el::Configurations::setGlobally(el::ConfigurationType, std::string const&, bool)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::info_container::register_info(rs2_camera_info, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::hid_sensor::get_custom_report_data(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, librealsense::platform::custom_sensor_report_field) const'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::info_container::get_info[abi:cxx11](rs2_camera_info) const'
../../librealsense2.so.2.8.2: undefined reference to `rosbag::Bag::appendHeaderToBuffer(rosbag::Buffer&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::sensor_base::get_info[abi:cxx11](rs2_camera_info) const'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::pipeline_config::enable_device(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::hw_monitor::get_firmware_version_string[abi:cxx11](int, unsigned int) const'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::hw_monitor::get_module_serial_string[abi:cxx11](unsigned char, unsigned int) const'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::pipeline_config::enable_record_to_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::pipeline_config::enable_device_from_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../librealsense2.so.2.8.2: undefined reference to `rosbag::Bag::writeHeader(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::uvc_sensor::uvc_sensor(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::shared_ptr<librealsense::platform::uvc_device>, std::unique_ptr<librealsense::frame_timestamp_reader, std::default_delete<librealsense::frame_timestamp_reader> >, librealsense::device*)'
../../librealsense2.so.2.8.2: undefined reference to `rosbag::Bag::open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)'
../../librealsense2.so.2.8.2: undefined reference to `el::Configurations::set(el::Level, el::ConfigurationType, std::string const&)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::playback_device::get_file_name() const'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::recoverable_exception::recoverable_exception(std::string const&, rs2_exception_type)'
../../librealsense2.so.2.8.2: undefined reference to `librealsense::hid_sensor::hid_sensor(std::shared_ptr<librealsense::platform::hid_device>, std::unique_ptr<librealsense::frame_timestamp_reader, std::default_delete<librealsense::frame_timestamp_reader> >, std::unique_ptr<librealsense::frame_timestamp_reader, std::default_delete<librealsense::frame_timestamp_reader> >, std::map<rs2_stream, std::map<unsigned int, unsigned int, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, unsigned int> > >, std::less<rs2_stream>, std::allocator<std::pair<rs2_stream const, std::map<unsigned int, unsigned int, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, unsigne [... cut ...]
../../librealsense2.so.2.8.2: undefined reference to `librealsense::context::remove_device(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
examples/capture/CMakeFiles/rs-capture.dir/build.make:99: recipe for target 'examples/capture/rs-capture' failed
make[2]: *** [examples/capture/rs-capture] Error 1
CMakeFiles/Makefile2:252: recipe for target 'examples/capture/CMakeFiles/rs-capture.dir/all' failed
make[1]: *** [examples/capture/CMakeFiles/rs-capture.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

I am using Ubuntu 16.04, 4.10.0-40-generic.

dorodnic, 2018-01-13:
Please make sure you follow the steps in [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) and open new issue if you still need help

closed 2018-01-13 by dorodnic