realsenseai/librealsense #855: No macOS SDK version

moorage, 2017-11-27:
Enhancement request: create a build for macOS!

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | Don't know  | 
| Operating System & Version |  10.13.1 + |

dorodnic, 2017-11-28:
Hi @moorage
If you clone the [development](https://github.com/IntelRealSense/librealsense/tree/development) branch right now, and build it from source using the existing [instructions](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_osx.md) you will get 99% of SDK functionality already. We are working to continuously improve and expand OSX support, and planning to address existing file import/export issues, 4K display issues and packaged distribution. 
Please let us know if you encounter any problems.

dorodnic, 2018-01-21:
The Mac OS version is mostly feature complete at this point and is in line with rest of the code-base (on the master branch). There are some known issues we are trying to address, but they are inherent to the platform (libuvc backend)

closed 2018-01-21 by dorodnic