realsenseai/librealsense #856: High density depth image

Panepo, 2017-11-28:
| Required Info  |   |
|---|---|
| Camera Model | D410 | 
| Firmware Version |  05.06.09.00 | 
| Operating System & Version |  Win10 (Build 16299) |
| Kernel Version (Linux Only)  |   |

Hi there,

There is an option in RS viewer named "Visual Preset";
When set this option to "High Density", the RS outputs a high density depth image.
How can I get this high density depth image in python?

Thanks

dorodnic, 2017-11-28:
Hi @Panepo 
One way to achieve this would be to export your desired settings from the viewer into a JSON file and later [load it from python](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py#L80).
This should copy over all relevant depth settings including the preset. 
Another approach would be to control the `RS2_OPTION_VISUAL_PRESET` option from python, but I can't find an existing code sample for that. 
Let us know if this helps, and we will add more example code if needed.

danielw256, 2017-11-29:
@Panepo Do you have Creators Update? If so, how did you get librealsense to work? I can get a RGB stream, but no depth stream.

Panepo, 2017-11-29:
Hi @dorodnic 

I have found the settings of the visual preset of viewer at [here](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/advanced_mode/presets.cpp), and [load it from python](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py).
but the output of the depth image is the same as before, is there something wrong on [my code](https://gist.github.com/Panepo/b554e9863b0cfa7a4b84d280afc9ea60)?
It looks like the device uses default settings, how to set the advance mode to RS correctly?

Thanks

Panepo, 2017-11-29:
@danielw256 
I have Creators Update and I follow this installation [guide](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/readme.md).
However my device is RS410, which doesn't have a RGB camera, but IR and depth stream works correctly.

dorodnic, 2017-11-29:
Hi @Panepo 
I have modified [opencv_viewer_example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py) to cycle through presets (similar to what the Viewer is doing):
```python
## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

###############################################
##      Open CV and Numpy integration        ##
###############################################

import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
cfg = pipeline.start(config)
dev = cfg.get_device()
depth_sensor = dev.first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 4)

iteration = 0;
preset = 0;
preset_name = '';

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        
        iteration = iteration + 1
        if iteration > 100:
            preset = preset + 1
            iteration = 0
            range = depth_sensor.get_option_range(rs.option.visual_preset)
            preset = preset % range.max
            depth_sensor.set_option(rs.option.visual_preset, preset)
            preset_name = depth_sensor.get_option_value_description(rs.option.visual_preset, preset)
        
        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, None, 0.5, 0), cv2.COLORMAP_JET)

        # Stack both images horizontally
        images = np.hstack((color_image, depth_colormap))
        
        font = cv2.FONT_HERSHEY_SIMPLEX
        cv2.putText(images, preset_name,(60,80), font, 4,(255,255,255),2,cv2.LINE_AA)

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()
```
In addition, `load_json` should also work. After reviewing your code, I had two ideas:
1. Instead of using `find_device_that_supports_advanced_mode`, get the device from pipeline configuration after start:
```python
cfg = pipeline.start(config)
dev = cfg.get_device()
```
2. Perform `load_json` after you start streaming

Panepo, 2017-11-30:
Hi @dorodnic 
Thanks a lot, it's works.

closed 2017-11-30 by Panepo

ValerioBiscione, 2018-04-05:
I just solved this as well and I thought of giving my code to the community.
Note that the json file saved with the rs-depth-quality **does not** save the info relative to resolution and fps. However, you can change them manually (they are just at the bottom) and then, in the script, read them from the json file and change it when calling the "config.enable_stream" function, as usual.

`lass RealSenseInterfaceAdvanced(RealSenseInterface):
	def __init__(self, jsonFile):
		self.DS5_product_ids = ["0AD1", "0AD2", "0AD3", "0AD4", "0AD5", "0AF6", "0AFE", "0AFF", "0B00", "0B01", "0B03", "0B07"]

		self.jsonFile = jsonFile;
		# Configure depth and color streams
		self.pipeline = rs.pipeline()
		self.config = rs.config()

		jsonObj = json.load(open(self.jsonFile))
		self.json_string = str(jsonObj).replace("'", '\"')

		print("Configuration " + self.jsonFile + " loaded");
		self.enablePipeline(w=int(jsonObj['stream-width']), h=int(jsonObj['stream-height']), freq=int(jsonObj['stream-fps']));

	def startStreaming(self):
		self.cfg = self.pipeline.start(self.config)
		cv2.waitKey(500)
		self.loadConfiguration();

	def loadConfiguration(self):
		self.dev = self.cfg.get_device()
		self.advnc_mode = rs.rs400_advanced_mode(self.dev)

		print("Advanced mode is", "enabled" if self.advnc_mode.is_enabled() else "disabled")
		self.advnc_mode.load_json(self.json_string)
	def autocycleStream(self):
		while (True):
			frames = self.pipeline.wait_for_frames()
			depth_frame = frames.get_depth_frame()
			# color_frame = frames.get_color_frame()
			if not depth_frame:
				continue

			# Convert images to numpy arrays
			depth_image = np.asanyarray(depth_frame.get_data())
			depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
			cv2.imshow('autocycleStream',depth_colormap)
			cv2.waitKey(0)

`

Then call

camera=RealSenseInterfaceAdvanced(jsonFile='configCamera.json');

camera.startStreaming()
camera.autocycleStream()

ColinSmith123, 2019-05-13:
this is what I currently have... there is no error, but the image isn't changing so I guess the preset isn't loaded correctly... 

import pyrealsense2 as rs
import numpy as np
import cv2
import imageio

import json

jsonFile = 'presets.json';

pipeline = rs.pipeline()
config = rs.config()

jsonObj = json.load(open(jsonFile))
json_string = str(jsonObj).replace("'", '\"')

print("Configuration " + jsonFile + " loaded");

cfg = pipeline.start(config)
cv2.waitKey(500)

dev = cfg.get_device()
advnc_mode = rs.rs400_advanced_mode(dev)
print("Advanced mode is", "enabled" if advnc_mode.is_enabled() else "disabled")
advnc_mode.load_json(json_string)

#config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)   
#config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)   
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

#enablePipeline(w=int(jsonObj['stream-width']), h=int(jsonObj['stream-height']), freq=int(jsonObj['stream-fps']));

depth_sensor = cfg.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

align_to = rs.stream.color
align = rs.align(align_to)

for i in range(1000000):
        frames = pipeline.wait_for_frames()
       
        aligned_frames = align.process(frames)       
        aligned_depth_frame = aligned_frames.get_depth_frame() 
        color_frame = aligned_frames.get_color_frame()
        
        if not aligned_depth_frame or not color_frame:
            continue
        
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, depth_colormap))
        
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        key = cv2.waitKey(1)
        
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
            
#        cv2.imwrite("../data/003/depth%d.jpg" % i, depth_colormap)
#        cv2.imwrite("../data/003/rgb%d.jpg" % i, color_image)
#               
else:       
    cv2.destroyAllWindows()
    pipeline.stop()

What it should look like: 
[image]

What it actually looks like: 
[image]

so if you have any tipps please send them to me :)