realsenseai/librealsense #857: Lost of frames during RealSense Viewer recording

mjfoo21, 2017-11-28:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.8.0.58  |

I tried to record the camera streams (RGB and depth) with the RealSense Viewer. Some of the depth frames were lost from time to time and the playback froze when this happened.

I have installed the SDK using installation packages in dpkg format following this [link](https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md).

During the installation, an error occurs when I run `sudo apt-get install realsense-uvcvideo` , which gives me:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  realsense-uvcvideo
0 upgraded, 1 newly installed, 0 to remove and 6 not upgraded.
Need to get 0 B/79.0 kB of archives.
After this operation, 814 kB of additional disk space will be used.
Selecting previously unselected package realsense-uvcvideo.
(Reading database ... 416891 files and directories currently installed.)
Preparing to unpack .../realsense-uvcvideo_1.1.0-0ubuntu3_all.deb ...
Unpacking realsense-uvcvideo (1.1.0-0ubuntu3) ...
Setting up realsense-uvcvideo (1.1.0-0ubuntu3) ...
Loading new realsense-uvcvideo-4.4.0 DKMS files...
First Installation: checking all kernels...
Building for 4.8.0-58-generic and 4.10.0-40-generic
Building initial module for 4.8.0-58-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/realsense-uvcvideo.0.crash'
Error! Bad return status for module build on kernel: 4.8.0-58-generic (x86_64)
Consult /var/lib/dkms/realsense-uvcvideo/4.4.0/build/make.log for more information.
Loading new realsense-uvcvideo-4.10.0 DKMS files...
First Installation: checking all kernels...
Building for 4.8.0-58-generic and 4.10.0-40-generic
Building initial module for 4.8.0-58-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/realsense-uvcvideo.0.crash'
Error! Bad return status for module build on kernel: 4.8.0-58-generic (x86_64)
Consult /var/lib/dkms/realsense-uvcvideo/4.10.0/build/make.log for more information.
```
Running the same command again gives:
```
realsense-uvcvideo is already the newest version (1.1.0-0ubuntu3). 
0 upgraded, 0 newly installed, 0 to remove and 171 not upgraded. 
```
Results from `modinfo uvcvideo | grep "version:"`:
```
version:        1.1.1
srcversion:     533BB7E5866E52F63B9ACCB
```
As discussed in [this issue](https://github.com/IntelRealSense/librealsense/issues/759), I tried removing and reinstalling the patch again but the results are the same.

I am wondering if this leads to the frame loss in the recorder or there are other bugs in the Viewer? I intend to use the recorder and playback in my c++ code for some testing.

I have previously installed librealsense-1 if it matters.

ev-mp, 2017-11-29:
Hello @mjfoo21,

- Did you verify that you had no previous realsense DKMS installed prior to this "realsense-uvcvideo" package ? As you've mentioned that you did/do have Librealsense1 installed as well. 
Trying to remove all librealsense-related  DKMS packages and reinstalling the latest version would be the first thing I would try. ("dkms status" will provide a list of all dkms-es present).
- The librealsense DKMS and the kernel patch do the same thing, so while it is not clear which patch you were trying to apply you should have either DKMS or the kernel patch applied but not both.
- The 4.8.0-xx versions of LTS were deprecated by Ubuntu and replaced by 4.10.0-xx so it will take time to try and reproduce the environment if indeed . If you are positive that this was a clean install I would suggest two alternative solutions:
  - Switch to 4.10.. kernel that you have installed and apply DKMS only. or,
  - Stick to 4.8... kernel, remove DKMS and manually apply the kernel patch as described  in the [installation manual](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-installation). Note that you only need to perform the linux backend preparation, and you can still use all the other librealsense debians:  udev-rules, pre-compiled demos and sdk.

mjfoo21, 2017-11-29:
Hello @ev-mp,

Thanks for the reply.
`dkms status` gives me
```
realsense-uvcvideo, 4.10.0: added
realsense-uvcvideo, 4.4.0: added
```
Sorry I am not very familiar with the commands on Linux. Can you show me how can I remove the DKMS packages and applying them again?

I have tried to run the cpp-capture in 4.10. The IR is not showing up and the depth is flickering with purplish hue. That's why I am sticking with 4.8 for now.

ev-mp, 2017-11-29:
Hi @mjfoo21 ,
The flickering depth channel is a symptom of a missing kernel patch, and installing the kernel patch manually  or via DKMS package is documented in the installation guide - (~15 min to do it).
I suggest you perform the following:
- Start with removing all librealsense packages from the 4.8 kernel: 'dpkg --list | grep -i "realsense" | cut -d " " -f 3 | xargs sudo dpkg –purge' (note that this will remove librealsense1 packages as well, if you used them)
- Switch to 4.10 and proceed according to the installation guide:
   - If you build librealsense from sources then follow the installation guide and invoke the kernel patch manually (see my previous response)
    - If you just want to run the binaries then use the [debians installers we provide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#linux-distribution).

mjfoo21, 2017-11-29:
Hello @ev-mp,

Is it possible to retain librealsense1 since I am using them as well?

ev-mp, 2017-11-29:
Absolutely, the two versions can and will co-exist.
Two notes though:
- You should use the udev rules and the kernel patch/dkms from librealsense2, which also cover the stream formats and devices supported by librealsense1.
- Make sure to use the proper include files as the names haven't changes, only the install paths:  ../include/librealsense**2**/rs.h vs ../include/librealsense**2**/rs.h.

mjfoo21, 2017-11-29:
Sorry I am a little bit confused. Just to make sure:
- I do not need to remove any librealsense packages from 4.8 since I need the first version
- Switch to 4.10 and clone librealsense2 to the system and following all the instructions in the installation guide, with everything built from the librealsense**2** directory
- The data from librealsense will be left untouched

ev-mp, 2017-11-29:
1. You don't need to uninstall/remove librealsense1 from 4.8, but you do need to uninstall DKMS package that was distributed with librealsense1 as it would collide with DKMS/kernel patch for librealsense2.
Again, you can use LRS1 and LRS2 simultaneously, but if you install both you need to set the kernel patches and udev rules distributed with LRS2.
2. Correct.
3. Correct, but remember to reapply the kernel patch/udev rules when switching back to kernel 4.8

mjfoo21, 2017-11-29:
I have removed dkms in 4.8 using `sudo dkms remove realsense-uvcvideo/4.4.0 --all` and `sudo dkms remove realsense-uvcvideo/4.10.0 --all`.
Then I switched to 4.10 and followed the installation instructions until "Recompile and install librealsense binaries". When I do `make`, I got this error:
```
/home/mj/librealsense2/librealsense/src/rs.cpp: In function ‘rs2_device* rs2_create_device_from_sensor(const rs2_sensor*, rs2_error**)’:
/home/mj/librealsense2/librealsense/src/rs.cpp:1636:44: error: could not convert ‘sensor->rs2_sensor::parent’ from ‘const rs2_device’ to ‘std::shared_ptr<librealsense::context>’
     return new rs2_device { sensor->parent };
                                            ^
CMakeFiles/realsense2.dir/build.make:446: recipe for target 'CMakeFiles/realsense2.dir/src/rs.cpp.o' failed
make[2]: *** [CMakeFiles/realsense2.dir/src/rs.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```
What is going wrong over here?

ev-mp, 2017-12-04:
Hi @mjfoo21 , sorry for a long delay.
Which librealsense branch you're building from ? 
Both 'master' and 'development' branches have CI integrated that so it is more likely that the local build/repo is broken. Another possibility is that you have librealsense2 include files from another version installed.
Try a fresh build against one of these branches and update.

ev-mp, 2017-12-13:
@mjfoo21 , do you still need assistance with this ?

mjfoo21, 2017-12-14:
Hello @ev-mp, sorry for the long time of inactivity.

I got it working now. Previously I cloned it from the master branch. After switching to the development branch everything installed smoothly.

Thanks a lot for the great help!

closed 2017-12-14 by mjfoo21