realsenseai/librealsense #858: Align Color to Depth

mjfoo21, 2017-11-28:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.8.0.58  |

Can `rs2::align` align depth to other streams? The [example](https://github.com/IntelRealSense/librealsense/tree/master/examples/align) states that 

> A `rs2::align` object always transforms depth images to some target image

When I do
```
rs2::align align(RS2_STREAM_DEPTH);
frame = pipe.wait_for_frames();
rs2::frameset proccessed = align.proccess(frame);
rs2::video_frame color_processed = proccessed.first(RS2_STREAM_COLOR);
```
it gives the error `Frame of requested stream type was not found!`.

Is there any way to generate `color_aligned_to_depth` stream like librealsense-1 in Version 2?

dorodnic, 2017-11-29:
Hi @mjfoo21 
I believe it's a bug. We will look into it and update this thread.

zivsha, 2017-12-06:
@mjfoo21 , I'm closing the issue since we pushed the fix to `development`, please let us know if this solved it for you or not.

closed 2017-12-06 by zivsha

mjfoo21, 2017-12-14:
Hi @zivsha,

I got the aligned-to-depth function. Thanks for fixing the issue!