realsenseai/librealsense #859: Not detecting ZR300 on Intel Aero hardware USB OTG issue

ChrisHughes, 2017-11-28:
Camera: ZR300
Firmware: Unknown (as shipped out of the box)
Ubuntu 16.04 LTS
Kernel 4.4
Librealsense 1.12.1

ZR300 connects and operates as expected on Ubuntu 16.04 on a laptop, however on the same OS, kernel on the Intel Aero Compute board, the camera is not detected (a small number of messages in dmesg, I'll post along with firmware when I'm back in the office on Thursday).

I am using a USB3.0 OTG cable to connect the camera, and have tried over both a powered USB hub as well as directly connected to the USB port. All other USB 3.0 devices tried function as expected.

Given the same OS, kernel and librealsense work on a laptop, my suspicions fall on the USB controller configuration in the InsydeH2O EFI bios. I'm unfamiliar with the configuration of this BIOS and have posted a screenshot in the hopes that someone might know what configuration options would help in establishing a working connection to the device.

[image]

Any help on the issue would be appreciated.

freemanlo, 2017-11-30:
Hi @ChrisHughes 
Please refer to below chapter 3.5 about RealSense USB 3.0 Camera Connector on Intel Aero board. This should be able to resolve your issue.
https://www.intel.com/content/dam/support/us/en/documents/boardsandkits/aero/intel-aero-compute-board-guide.pdf

dorodnic, 2017-12-05:
@ChrisHughes please let us know if it helped, and re-open the issue if you need any further assistance.

closed 2017-12-05 by dorodnic

ChrisHughes, 2017-12-06:
Hi @freemanlo

The ZR-300 does not come with a 10-pin IPEX connector as the R-200 did. I am therefore trying to connect this via the USB 3.0 OTG port on the side of the Intel Atom. While I have the correct cables to connect this, it does not appear to be detecting.

I do not have the facility to create a custom IPEX to USB 3.0 OTG plug as I am on the road currently but need this up and running. Is there any BIOS settings I can modify so that the device is detected via this means of connecting it?

ChrisHughes, 2017-12-11:
I've provided some updated info here, this is still an issue. Can you please reopen this.