realsenseai/librealsense #860: get IR frame - Python

cferns, 2017-11-28:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  SDK2.0  | 
| Operating System & Version |  Windows 10  |

I am using the 'Python' wrapper. I was not able to figure out how to get the Infrared (IR) stream/frame from the camera. There are 'get_depth_frame' and 'get_color_frame" calls to get the depth and the color stream but none for the IR stream.

danielw256, 2017-11-29:
@cferns Do you have Creators Update? If so, how did you get librealsense to work? I can get a RGB stream, but no depth stream.

MichaelHanssonPH, 2017-11-29:
@cferns I asked myself the same question a few days ago; why is there no "get_IR_frame()" call? But I finally figured it out and the following code works for me (using Python 3.6 and OpenCV 3.3), in this example to stream and display an IR image until the user hits ESC:

```
import pyrealsense2 as rs
import numpy as np
import cv2

# Configure IR stream
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30)

# Start input streaming
pipeline.start(config)

# Ignore first 1sec for camera warm-up
for i in range(30):
    frames = pipeline.wait_for_frames()

try:
    while True:

        # Read image
        frames = pipeline.wait_for_frames()
        infrared_frame = frames.first(rs.stream.infrared)
        IR_image = np.asanyarray(infrared_frame.get_data())

        # Display image
        cv2.imshow('IR image', IR_image)

        # Exit on ESC key
        c = cv2.waitKey(1) % 0x100
        if c == 27:
            break

finally:
    pipeline.stop() 
    cv2.destroyAllWindows()

```

cferns, 2017-11-29:
@danielw256 I do not have the creators update. My current windows version is 10.1607 on which I get the RGB and depth data as well.

cferns, 2017-11-29:
@MichaelHanssonPH Thank you very much. I was not able to enable the IR stream due to incorrect syntax. Thanks again.

closed 2017-11-30 by cferns