realsenseai/librealsense #861: Where is pyrealsense2.py after compilation?

felix-tracxpoint, 2017-11-29:
| Required Info  |   |
|---|---|
| Camera Model |  D400 | 
| Firmware Version |  Don't have the sensor on hand right now... | 
| Operating System & Version |  Windows 10 |
| Kernel Version (Linux Only)  |   |

I would like to use the Python wrapper and I think that I managed to compile it (what I did was run Cmake_gui which created a **c:\realsense\librealsense-master\build\wrappers\python\RealsensePythonWrappers.sln** solution file that I promptly opened and built without errors). 

But where is pyrealsense2.py, the creation of which I had assumed was the point of the whole compilation procedure? I cannot find any .py files un the whole build directory.

Am I missing something really obvious? Please pardon my utter noobishness.

dorodnic, 2017-11-29:
Open `librealsense2.sln` under build directory, build `pyrealsense2` project under `Wrappers` solution-folder, `pyrealsense2.pyd` will be under `build\Release\pyrealsense2.pyd` or `build\Debug\pyrealsense2.pyd`.
Let us know if this helps

dorodnic, 2017-11-29:
You can also use pre-compiled `pyrealsense2.pyd` provided with the [installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.8.2/Intel.RealSense.SDK.exe).

felix-tracxpoint, 2017-11-29:
So, in the meantime I figured out that there is no .py file but rather the .pyd file is the one to work with. I haven't found the pre-compiled one but the one I compiled works fine. Many thanks for getting back to me.

closed 2017-11-29 by ChrisH-Intel